arduino/Arduino #10914: [Library Manager] Request to add new library

ansonhe97, 2020-11-02:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

Hi,

Please add the following to the Library Manager: 

- **Seeed_Arduino_LIS3DHTR**
https://github.com/Seeed-Studio/Seeed_Arduino_LIS3DHTR

Many Thanks

per1234, 2020-11-02:
Hi @ansonhe97. Thanks so much for your interest in contributing to the Arduino Library Manager index!

The library.properties `name` value of your library is not compliant with the requirements for addition to the Library Manager index due to the use of disallowed characters (`±`, `(`, `)`).

You can see the allowed characters here:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

So please fix the `name` value and then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

ansonhe97, 2020-11-02:
HI @per1234 

Thanks for the swiftly reply, please check the library again 👍

per1234, 2020-11-02:
I'm sorry to say that the `name` value still contains the disallowed parentheses characters.

Please resolve the issue, do the release/tag, and comment here.

ansonhe97, 2020-11-02:
Please check again, thanks

ansonhe97, 2020-11-04:
Hi @per1234 

Wondering how long does this process normally take?

per1234, 2020-11-04:
Hello. Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

ansonhe97, 2020-11-05:
Thanks for this information, much appreciated !

cmaglie, 2020-11-06:
Done.

closed 2020-11-06 by cmaglie