arduino/Arduino #10919: [Library Manager] Please add LinearRegression library to manager.

akkoyun, 2020-11-04:
Hi all,
Can u add my library to library manager.

https://github.com/akkoyun/LinearRegression

per1234, 2020-11-04:
Hi @akkoyun. In order to be added to the Library Manager index, you must have a `url` value defined in your library.properties. I It is currently undefined:
https://github.com/akkoyun/LinearRegression/blob/01.00.00/library.properties#L8
```
url=
```
You can just add the URL of your repository if you like.

After fixing that, please either update the `version` value in library.properties and create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

akkoyun, 2020-11-04:
Updated please confirm
thanks.

cmaglie, 2020-11-06:
Done.

closed 2020-11-06 by cmaglie

akkoyun, 2020-11-06:
Thanks my friend.

njh, 2021-11-30:
It looks like this repository has now been deleted.

Do you mind giving some feedback about why it is gone @akkoyun ?