arduino/Arduino #10921: arduino IDE fonts too small to read

Sasawaws, 2020-11-04:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

Sasawaws, 2020-11-04:
Since "upgrading" to kubuntu20.04 the arduino IDE starts with such ridiculously small menu fonts (4pt?) it is unusable. 
Obviously any menus that might influence the font size are also unusable if they exist.  Nothing in the system fonts appears to fix this.  All other applications seem fine.  Same result 1.8.9 and 1.8.13.  It's quite annoying, I'm sure there is a simple fix but it eludes me (and clearly shouldn't be the case).

Sasawaws, 2020-11-04:
I forgot to add the default font size for the main panel was similarly tiny but ctrl-scroll button zoomed this to "reasonable"

Sasawaws, 2020-11-06:
After endless trying I changed my monitor and then font is not readable - very small but readable... All other applications remained roughly the same so I don't know what arduino is up to but right now it's working for me...

BackMountainDevil, 2021-06-04:
[Here](https://forum.arduino.cc/t/how-to-change-arduino-ide-menu-font-size/138861/16) are three ways to change the menu fontsize. Those do help someone. But no one works for me...