arduino/Arduino #10925: Cloudflare Error 522

RonNovy, 2020-11-05:
Getting lots of random Cloudflare Error 522 on links in the reference section of arduino.cc

These links do not currently work for me:
https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/toint
Gives: Error 522 Ray ID: 5ed92dc988df0530

https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/indexof
Gives: Error 522 Ray ID: 5ed93f5b4b6e0530

Using a proxy I can see that they sometimes do work through the proxy but other times they do not.
Some links that were broken are now working.  Like this link didn't work at first but now works:
https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/lastindexof/

Not sure what is happening but thought I'd report it.

RonNovy, 2020-11-06:
I believe I've discovered a fix.  The links above will work if you add a slash ('/') to the end of the URL.

This works:
https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/toint/

but without the slash at the end it does not.  It seems the broken links are the ones without the slash at the end.

endorama, 2020-11-06:
Hi @RonNovy, I'm Edoardo, from the Operations team at Arduino. We have been subjected to a DDOS attack this week and the unfortunate consequence of it has been some pages going offline for some time. We enabled mitigations and were able to deflect part of it, but occasionally some page went offline. This happened mainly Tuesday and Wednesday.
Everything should be back to normal now. Thanks for reporting and sorry for the troubles.

I'm going to close this, feel free to get in touch again if you see the same behaviour happening.

dbuggz, 2020-11-08:
Still happening in Australia:
Cloudfare error:
Error 522 Ray ID: 5ef2409e2c3216a9 • 2020-11-08 21:06:30 UTC

Dave

endorama, 2020-11-09:
Hello, I'm going to reopen this as we have received further reports of this error.

For everyone posting in this issue, please include:
- the URL you were trying to visit
- the Ray ID
- the Timestamp

This will help the team spot any inconsistent behaviour.

We started working on it already and identified a possible root cause.

dbuggz, 2020-11-09:
Hi, just tried
https://www.arduino.cc/en/Main/Products
https://www.arduino.cc/en/Guide/NANO33BLESense
https://www.arduino.cc/en/Reference/WiFiNINA

All three OK here, Newcastle Australia, 2020-11-09 20:56 (Approx) UTC

Thank you
Dave

endorama, 2020-11-10:
I believe we found the root cause and resolved it by yesterday. Our observations and multiple reports confirm this has not been resolved. Thank you for your patience.

I'm closing this again, please report any further issue. Thank you!

closed 2020-11-10 by endorama