arduino/Arduino #10927: [Library Manager] Please add tinyhal Arduino Library

lexus2k, 2020-11-05:
Greetings.

Could you please add tinyhal library to Arduino Library Manager?
https://github.com/lexus2k/tinyhal

This library is basis for future tinyproto library versions and some other libraries.

Thank you

per1234, 2020-11-06:
Hi @lexus2k. Thanks for your interest in contributing to the Arduino Library Manager index!

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your `library.properties` file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

lexus2k, 2020-11-06:
I'm so sorry. I've completely forgotten about that. 
Fixed, added release v0.1.0

cmaglie, 2020-11-06:
Done.

closed 2020-11-06 by cmaglie