arduino/Arduino #10930: [Library Manager] Please add the CanAirIO Air Quality Sensors library v0.1.2

hpsaturn, 2020-11-06:
Greetings.

Could you please add **CanAirIO AirQuality Sensors Library** to Arduino Library Manager?

- [x] Github repository: https://github.com/kike-canaries/canairio_sensorlib
- [x] tag: **v0.1.2**
- [x] library.properties
- [x]  library is compliant with 1.5 format
- [x] Arduino sample
- [x] **arduino-cli** documentation ready

cmaglie, 2020-11-06:
hi @hpsaturn,
I've added the library to the indexer but there is a problem, you must define the `author` field in library.properties.

I suggest the following procedure:

1. add `author` field in library.properties
2. commit & push
3. delete the 0.1.2 tag and redo it on the new `master`

since the library is already registered in our indexer, once the steps above are done it will be automatically published within an hour. Please comment here if you need further assistance.

closed 2020-11-06 by cmaglie

hpsaturn, 2020-11-06:
Ouch! 

Sorry for my mistake. We are ready. Please review the change that I did.

Thanks!