arduino/Arduino #10931: Overwriting a ino file with "Save as" deletes all data in folder

hgjersdal, 2020-11-06:
Hello
I opened a new project in Arduino IDE 1.8.13 on Arch linux, wrote some code in it and saved it to the default location, ending up with a project called something like sketch_nov06b.ino. 
I wanted to overwrite an old project I had, as this code did the same job but better. I clicked "File -> Save As...", and saved it as a dot ino file that already existed. I was prompted that this would delete the old dot ino file, which I was ok with. In addition to deleting the dot ino file, it deleted the whole folder, with all the data in it, and created a new folder with the same name containing nothing but the new ino file. 

# Steps to reproduce
Have a folder with a sketch and some important data.

    $ pwd
    /home/haavagj/git/test_project
    $ ls
    important_data.txt  important_data.txt~  test_project.ino  test_project.ino~ 
    $ cat test_project.ino
    void setup() {
      // put your setup code here, to run once:
      unimportand_code();
    }
    
    void loop() {
      // put your main code here, to run repeatedly:
      more_unimportant_code();
    }

Overwrite test_project.ino with Arduino IDE 1.8.13 on Arch linux and lose important data.

    $ ls
    ls: reading directory '.': No such file or directory
    $ pwd
    /home/haavagj/git/test_project
    $ cd ../test_project 
    $ ls
    test_project.ino
    $ cat test_project.ino 
    void setup() {
      // put your setup code here, to run once:
      improved_setup();
    }
    
    void loop() {
      // put your main code here, to run repeatedly:
      improved_loop();
    }

per1234, 2020-11-06:
Related to the discussion in https://github.com/arduino/Arduino/issues/10879 regarding the IDE's save/save as UI not clearly communicating to the user that the sketch the the complete folder, not just the .ino file.

hgjersdal, 2020-11-06:
I do hope you understand that important_data.txt was a lot more important to me than what the Arduino definition of a sketch is. From the terminal, my text editor and python, the sketch looked a lot like a regular directory one could safely store things in.