arduino/Arduino #10935: [Bug] IDE adds additional characters when commenting out the code

rtek1000, 2020-11-08:
Hi,

I'm having problems with the Arduino IDE:
- When I open the IDE through the start menu, the IDE works normally,
- But after opening through the sketch (* .ino file), when using the shortcut Ctrl + / a, the IDE adds an additional bar.

The additional bar is inserted wherever the cursor is, if a block is selected to be commented, then the additional bar replaces the entire block.

Examples of the problem:

Initial code:
```C++
void setup() {
   // put your setup code here, to run once:
 
 }
 
 void loop() {
   // put your main code here, to run repeatedly:
 
 }
```

Commenting with cursor before "void":

```C++
 ///void setup() {
   // put your setup code here, to run once:

 }
 
 void loop() {
   // put your main code here, to run repeatedly:
 
 }
```

Commenting with cursor before "setup":

```C++
 //void /setup() {
   // put your setup code here, to run once:

 }
 
 void loop() {
   // put your main code here, to run repeatedly:
 
 }
```

Commenting the "setup" routine block:

```C++
 ///
 
 void loop() {
   // put your main code here, to run repeatedly:
 
 }
```

O.S.: Linux Xubuntu:

$cat /etc/os-release

> NAME="Ubuntu"
> VERSION="20.04.1 LTS (Focal Fossa)"
> ID=ubuntu
> ID_LIKE=debian
> PRETTY_NAME="Ubuntu 20.04.1 LTS"
> VERSION_ID="20.04"
> HOME_URL="https://www.ubuntu.com/"
> SUPPORT_URL="https://help.ubuntu.com/"
> BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
> PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
> VERSION_CODENAME=focal
> UBUNTU_CODENAME=focal

per1234, 2020-11-08:
Thanks for your report @rtek1000!

Which version of the Arduino IDE are you using?

Did you install it from the official download on the Arduino IDE website (https://www.arduino.cc/en/Main/Software), or via a package manager (e.g., Snap, Brew, APT)?

rtek1000, 2020-11-08:
Hi,

IDE v.: 1.8.13,

I downloaded the linux package and installed it manually, via installer.

https://www.arduino.cc/en/Guide/Linux

Thank you.

per1234, 2021-03-29:
I see now that we already have a prior issue report about this at https://github.com/arduino/Arduino/issues/5464.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.
If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-03-29 by per1234