arduino/Arduino #10936: MacOS - compiles but won't run (error loading Java)

oobi, 2020-11-09:
I'm unsure how to resolve this. I've installed JDK8 per the instructions and I can get the thing to compile fine. The app launches but immediately shows an alert "Error loading Java" and quits. 

I've checked and rechecked the paths in the output and everything looks fine. As I understand it, the thing bundles the runtime environment into itself so shouldn't be reliant on whatever version of Java the system runs, is that correct?

ant build produces no error. 

ant run produces:

```
BUILD FAILED
/Users/ccarey/Sites/arduino-ide/Arduino/build/build.xml:142: The following error occurred while executing this line:
/Users/ccarey/Sites/arduino-ide/Arduino/build/build.xml:146: The following error occurred while executing this line:
/Users/ccarey/Sites/arduino-ide/Arduino/build/build.xml:543: exec returned: 1
```

When I run the app directly from its container I get this:
 ./macosx/work/Arduino.app/Contents/MacOS/Arduino

```
2020-11-09 10:56:50.141 Arduino[9480:281517] Loading Application 'Arduino'
2020-11-09 10:56:50.142 Arduino[9480:281517] JVMRuntime=adoptopenjdk-8.jdk
2020-11-09 10:56:50.142 Arduino[9480:281517] CFBundleName=Arduino
2020-11-09 10:56:50.142 Arduino[9480:281517] WorkingDirectory=(null)
2020-11-09 10:56:50.142 Arduino[9480:281517] JVMMainClassName=processing.app.Base
2020-11-09 10:56:50.142 Arduino[9480:281517] JVMOptions=(
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
2020-11-09 10:56:50.142 Arduino[9480:281517] JVMArguments=(
)
2020-11-09 10:56:50.143 Arduino[9480:281517] JVMClasspath=(null)
2020-11-09 10:56:50.143 Arduino[9480:281517] JVMDefaultOptions={
}
2020-11-09 10:56:50.143 Arduino[9480:281517] SearchSystemJVM=false
2020-11-09 10:56:50.143 Arduino[9480:281517] -> Bundle path: /Users/ccarey/Sites/arduino-ide/Arduino/build/macosx/work/Arduino.app
2020-11-09 10:56:50.143 Arduino[9480:281517] -> Working Directory: '/Users/ccarey/Sites/arduino-ide/Arduino/build'
2020-11-09 10:56:50.143 Arduino[9480:281517] -> JVM Runtime path: /Users/ccarey/Sites/arduino-ide/Arduino/build/macosx/work/Arduino.app/Contents/PlugIns/adoptopenjdk-8.jdk
2020-11-09 10:56:50.143 Arduino[9480:281517] -> Java Runtime Dylib Path: '/Users/ccarey/Sites/arduino-ide/Arduino/build/macosx/work/Arduino.app/Contents/PlugIns/adoptopenjdk-8.jdk/Contents/Home/jre/lib/jli/libjli.dylib'
```

Am I missing something obvious?

iiiidddd, 2020-11-12:
I'm having the same issue.

I notice if I run the official executable directly I get a different JVMRuntime string:

Official `JVMRuntime=jre8u252-b09.jre`
vs
Self-built `JVMRuntime=adoptopenjdk-8.jdk`

Not sure if that's just an alias and it's actually the same runtime, but I installed java using brew as suggested in [the wiki build guide](https://github.com/arduino/Arduino/wiki/Building-Arduino#mac-os-x).

```
$ java -version
openjdk version "1.8.0_272"
OpenJDK Runtime Environment (AdoptOpenJDK)(build 1.8.0_272-b10)
OpenJDK 64-Bit Server VM (AdoptOpenJDK)(build 25.272-b10, mixed mode)
```

ridgemcghee, 2020-12-10:
I'm having the same issue, too.
Same symptoms (output) as OP.
I'd appreciate any hints as to how to start debugging the issue.

jmartone, 2021-01-31:
I'm having the same issue myself, anyone crack it yet?

nklugman, 2021-07-21:
Bumped into this same issue (I think) 

`user@computer build % java -version
openjdk version "1.8.0_292"
OpenJDK Runtime Environment (AdoptOpenJDK)(build 1.8.0_292-b10)
OpenJDK 64-Bit Server VM (AdoptOpenJDK)(build 25.292-b10, mixed mode)`

JL03-Yue, 2023-01-19:
Exactly the same issue. Trying to build Arduino IDE from source code based on this [wiki](https://github.com/arduino/Arduino/wiki/Building-Arduino): ant dist, and ant build were successful, but ant start resulted in "error loading java" 
My java -version is:
`annie@ML-ART-03 build % java -version
openjdk version "1.8.0_292"
OpenJDK Runtime Environment (AdoptOpenJDK)(build 1.8.0_292-b10)
OpenJDK 64-Bit Server VM (AdoptOpenJDK)(build 25.292-b10, mixed mode)`