arduino/Arduino #10943: [Library Manager] Please add Pixetto library.

pixetto, 2020-11-10:
Please add Pixetto library to Library Manager.
[https://github.com/pixetto/Pixetto](https://github.com/pixetto/Pixetto)
Thanks.

per1234, 2020-11-10:
Hi @pixetto. Thanks for your interest in contributing to the Arduino Library Manager index!

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

pixetto, 2020-11-10:
I added a tag "1.4.0".
Please check it again.
I am sorry for my carelessness.

cmaglie, 2020-11-18:
Done.

closed 2020-11-18 by cmaglie