arduino/Arduino #10944: [Library Manager] Please add arduino-tone-pitch library

RodrigoDornelles, 2020-11-10:
Please add arduino-tone-pitch library to Library Manager.
https://github.com/rodrigodornelles/arduino-tone-pitch
Thanks.

RodrigoDornelles, 2020-11-14:
@per1234 the project should now conform to the standards of the Arduino Library Manager. 
https://github.com/RodrigoDornelles/arduino-tone-pitch/issues/1#issuecomment-727205804

per1234, 2020-11-14:
There is still one remaining issue:

Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

RodrigoDornelles, 2020-11-14:
Done!

cmaglie, 2020-11-18:
Done.

closed 2020-11-18 by cmaglie