arduino/Arduino #10946: [Bug] Auto formatting does not work if there is an 'R "===== (' in the sketch?

rtek1000, 2020-11-11:
Hi,

Auto formatting does not work if there are:
```C++
const char webSite1[] PROGMEM = R"=====(
"abc"
)=====";
```
But Auto formatting works if there is no line break:

```C++
const char webSite1[] PROGMEM = R"=====("abc")=====";
```

In both cases the compilation is successful.

Arduino IDE 1.8.13 (Linux)

per1234, 2020-11-11:
There is a bug report about it here in the repository of the Artistic Style tool used by the Arduino IDE's Auto Format feature: https://sourceforge.net/p/astyle/bugs/519/
It is claimed to be fixed (though there hasn't been a release of Artistic Style since that time).

rtek1000, 2020-11-11:
Interestingly, it may be better for the user to leave this type of variable in an additional file, such as index.h,

`#include "index.h"`

And thus can also use an online formatter to improve the reading of the code:

https://www.freeformatter.com/html-formatter.html

```
<html>
<body>
</body>
</html>
```
```
<html>
   <body></body>
</html>
```

rtek1000, 2020-11-11:
> There is a bug report about it here in the repository of the Artistic Style tool used by the Arduino IDE's Auto Format feature: https://sourceforge.net/p/astyle/bugs/519/
> It is claimed to be fixed (though there hasn't been a release of Artistic Style since that time).

I asked in the post when a new version of this software will be released, thanks!

roblatour, 2021-05-23:
I add that in other parts of the code, when this bug appears it puts extra spaces before / after the slash character "/" if they are surrounded by quotes.  For example "one/two" gets transformed to "one / two" even if is in a statement 50 lines away.

per1234, 2021-05-25:
Hi @rtek1000. There is some good news related to this bug. https://github.com/arduino/Arduino/pull/11543 will change the Arduino IDE from using the Artistic Style formatter to [ClangFormat](https://clang.llvm.org/docs/ClangFormat.html), which is able to handle raw string literals correctly.

ClangFormat is very popular and actively developed, and even suggested by the Artistic Style developer in [this statement](https://sourceforge.net/p/astyle/bugs/548/) where they state that they are no longer able to play as active a role in developing Artistic Style.

You can give the test build for the ClangFormat/Arduino IDE integration a try. The downloads are available here:
https://github.com/arduino/Arduino/pull/11543#issuecomment-845917365
If you do try it out, please let us know how it goes!