arduino/Arduino #10960: MKR 1000 can't be registered through the IoT site because the binary doesn't exist on Arduino's server

TimJAllen, 2020-11-15:
About a month ago I was able to register my MKR1000 board with the IOT dashboard and get a very simple test project running. I went to revisit it a few days ago and noticed that through the online editor the board was recognized, but the code wouldn't upload to it. I deleted the board from my IOT account and started the process of re-registering it. All 3 steps would complete, then the final screen would never finish and the board could not register.
I then bought the MKR1010 and tried to register, and the MKR1010 was able to register successfully.

After looking into the requests from my browser, it looks like the MKR1000 binary is missing from Arduino's servers.
The last call my browser is trying to make it to this URL before the MKR1000 can be registered is turning back a 404:
`https://api2.arduino.cc/iot/v2/binaries/ninaversion?fqbn=arduino:samd:mkr1000`

In Postman, I can run this intentionally wrong URL to get a list of valid fqbn's: 
`https://api2.arduino.cc/iot/v2/binaries/ninaversion?fqbn=THIS_IS_THE_WRONG_URL` 
and get this response:
```
{
    "id": "nb2AXVfB",
    "code": "invalid_request",
    "status": 400,
    "detail": "value of fqbn must be one of \"arduino:samd:mkr1000\", \"arduino:samd:mkrwifi1010\", \"arduino:samd:mkrgsm1400\", \"arduino:samd:nano_33_iot\", \"arduino:samd:mkrnb1500\", \"arduino:samd:mkrwan1300\", \"arduino:samd:mkrwan1310\", \"arduino-beta:mbed:envie_m7\" but got value \"THIS_IS_THE_WRONG_URL\"",
    "meta": {
        "attribute": "fqbn",
        "expected": "\"arduino:samd:mkr1000\", \"arduino:samd:mkrwifi1010\", \"arduino:samd:mkrgsm1400\", \"arduino:samd:nano_33_iot\", \"arduino:samd:mkrnb1500\", \"arduino:samd:mkrwan1300\", \"arduino:samd:mkrwan1310\", \"arduino-beta:mbed:envie_m7\"",
        "value": "THIS_IS_THE_WRONG_URL"
    }
}
```

So now that I know the site is running, it lists all the board available. I tried the URL for the MKR1010 and got this response
`https://api2.arduino.cc/iot/v2/binaries/ninaversion?fqbn=arduino:samd:mkrwifi1010`

```
{
    "bin": "AIAAIHEoAABZK...(MASSIVE BINARY BLOB)...AAGU5AABFSgAAtSAAAAAAAAAAAAAA",
    "filename": "ninaversion.arduino.samd.mkrwifi1010.bin",
    "fqbn": "arduino:samd:mkrwifi1010",
    "name": "ninaversion",
    "sha256": "2b766e27134d64834eba52c98de3c8f01d66edffa9163af52cb72952197e48d5"
}
```

So the API needed to register the devices is up. The only board I know of that isn't registerable at the moment is the MKR1000. 
I hope this is the right place to call attention to the issue.

per1234, 2020-11-15:
Thanks so much for the excellent investigation and report about this bug @TimJAllen!

One thing I'll add is that the MKR1000 doesn't use a NINA WiFi module, so, without any knowledge about this subject, I'd guess that the problem is not that the `https://api2.arduino.cc/iot/v2/binaries/ninaversion?fqbn=arduino:samd:mkr1000` resource is missing, but that it's being called at all.

The correct place to report issues related to Arduino Create is in the relevant section of the Arduino forum, in the case of Arduino IoT Cloud, here:
https://forum.arduino.cc/index.php?board=128.0
So I'll go ahead and close this.

I see there is already a report about the issue:
https://forum.arduino.cc/index.php?topic=713707
but without the explanation of the cause you provided, so I have replied there with a link to this issue so all the information is sure to be available to the Arduino Create team. Feel free to add any additional information.

closed 2020-11-15 by per1234

per1234, 2020-11-16:
Hi @TimJAllen. I believe the issue should be fixed now. Please give it a try and let us know if you have any problems.

TimJAllen, 2020-11-16:
@per1234 It works now! Thank you :)
It looks like the URL still returns a 404, but the registration goes through. Im assuming its because the Arduino no longer relies on that last call for set up?

Either way, thank you for the follow-up.

CMorooney, 2021-10-15:
> @per1234  I'm currently unable to set up my MKR1000 on the iot cloud (board has latest firmware)
> 404
> https://api2.arduino.cc/iot/v2/binaries/ninaversion?fqbn=arduino:samd:mkr1000

Resolved on forum
https://forum.arduino.cc/t/mkr1000-can-not-connect-to-iot-cloud-error-404/914827/5?u=calvinmorooney