arduino/Arduino #10963: speeding up the start through limiting the entries in preferences recent.sketches=

StefanL38, 2020-11-15:
Today I looked up the preferences.txt-file to examine what additional switches are hidden in this file
Then I saw that the recent.sketches= -item had 70kB of entries!!

If I click on file - "open recent"    the list shows ten files. Practical compromise between too much files and just three.

But why the heck does the IDE collect them all in the preference.txt-file?? Even if the 11 to 1000 entry are not accessible??
Can somebody give me a single reason why the Arduino-IDE can only work reliable with collecting all these entries?

That is what I call an old unvisible bug nobody wants to to clean up. 
But this bug slows down the startup of the IDE. I guess if the line of code is found that causes this 
adding the new file-entry and cut-away everything behind the tenth entry is done within 15 minutes or even less.

best regards Stefan

magedrifaat, 2021-01-24:
While I too think solving this is trivial, I tried increasing the entry to 1MB of paths, and decreasing it to 10KB and it had no noticeable impact on the startup of the IDE at least on my machine.

philbowles, 2021-05-14:
Well on MY machine (when some of the 65kb of paths no longer existed) it was taking anywhere between 3:30 and 7 minutes to start up. MINUTES. It needs fixing.

magedrifaat, 2021-05-15:
@philbowles What are the specs of the machine you are running? (i.e. what OS, RAM, how full is the HDD/SSD)
Also, did you see a faster start-up time when clearing the entries? Did you confirm that this is the real cause of your slow start-up time?

ednieuw, 2023-01-09:
Is this still an issue? Starting IDE 1.8.19 on a fast W11 PC with SSD took over 30 seconds.
Deleting most of the entries at: 
c:\Users\__username__\AppData\Local\Arduino15\preferences.txt 
after  recent.sketches=
speeded up starting significantly to 5 seconds.