arduino/Arduino #10964: Arduino 1.8.13 crashes on startup on macOS Big Sur 11.0.1

MenKuch, 2020-11-15:
Hi,. just installed the final version of macOS 11.0.1 Big Sur (20B29) and the Arduino IDE 1.8.13 crashes during launch. The startup window becomes visible and then it crashes with the following crashlog:

Process:               Arduino [892]
Path:                  /Applications/Arduino.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               1.8.13 (1.8.13)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [892]
User ID:               501

Date/Time:             2020-11-15 16:54:34.501 +0100
OS Version:            macOS 11.0.1 (20B29)
Report Version:        12
Bridge OS Version:     3.0 (14Y908)
Anonymous UUID:        FA0AC80F-0906-8AB2-61F8-5EC65B0DB62C

Time Awake Since Boot: 480 seconds

System Integrity Protection: enabled

Crashed Thread:        0  AppKit Thread  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00007fe095300000
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x7fe095300000:
    MALLOC_TINY              7fe095200000-7fe095300000 [ 1024K] rw-/rwx SM=PRV  
--> MALLOC_LARGE             7fe095300000-7fe095318000 [   96K] rw-/rwx SM=PRV  
    MALLOC_LARGE             7fe095318000-7fe095326000 [   56K] rw-/rwx SM=PRV  

Application Specific Information:
abort() called

Thread 0 Crashed:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff20312462 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff20340610 pthread_kill + 263
2   libsystem_c.dylib             	0x00007fff20293720 abort + 120
3   libjvm.dylib                  	0x000000010e4d72bb os::abort(bool) + 25
4   libjvm.dylib                  	0x000000010e5fe0e6 VMError::report_and_die() + 2304
5   libjvm.dylib                  	0x000000010e4d8ee6 JVM_handle_bsd_signal + 1131
6   libjvm.dylib                  	0x000000010e4d514b signalHandler(int, __siginfo*, void*) + 47
7   libsystem_platform.dylib      	0x00007fff20384d7d _sigtramp + 29
8   ???                           	0x00007fff884d7bf8 OBJC_CLASS_$_NSCalendar + 40
9   com.apple.Foundation          	0x00007fff211867ed NSCopyMemoryPages + 57
10  com.apple.Foundation          	0x00007fff2118ee59 -[NSConcreteMutableData appendBytes:length:] + 514
11  com.apple.Foundation          	0x00007fff211a5e8b -[NSData(NSData) enumerateByteRangesUsingBlock:] + 105
12  com.apple.Foundation          	0x00007fff21205d08 -[NSData(NSData) mutableCopyWithZone:] + 173
13  com.apple.AppKit              	0x00007fff22dc9549 __75-[NSBitmapImageRep _withoutChangingBackingPerformBlockUsingBackingCGImage:]_block_invoke + 339
14  com.apple.AppKit              	0x00007fff22dc93f0 -[NSBitmapImageRep _withoutChangingBackingPerformBlockUsingBackingCGImage:] + 77
15  com.apple.AppKit              	0x00007fff22dc9350 __53-[NSBitmapImageRep _performBlockUsingBackingCGImage:]_block_invoke + 116
16  com.apple.AppKit              	0x00007fff22dc9292 -[NSBitmapImageRep _performBlockUsingBackingCGImage:] + 77
17  com.apple.AppKit              	0x00007fff22dc9208 -[NSBitmapImageRep CGImage] + 122
18  com.apple.AppKit              	0x00007fff22d975fe __48-[NSImage CGImageForProposedRect:context:hints:]_block_invoke + 91
19  com.apple.AppKit              	0x00007fff22d9724e -[NSImage _usingBestRepresentationForRect:context:hints:body:] + 129
20  com.apple.AppKit              	0x00007fff22d96b1e -[NSImage CGImageForProposedRect:context:hints:] + 770
21  com.apple.AppKit              	0x00007fff22dc5394 -[_NSSimpleImageView updateLayer] + 1190
22  com.apple.AppKit              	0x00007fff22d2ad2c _NSViewUpdateLayer + 98
23  com.apple.AppKit              	0x00007fff22c3ce28 +[NSAppearance _performWithCurrentAppearance:usingBlock:] + 66
24  com.apple.AppKit              	0x00007fff22e9d235 __30-[_NSViewBackingLayer display]_block_invoke + 113
25  com.apple.AppKit              	0x00007fff22c7bbe1 -[NSFocusStack performWithFocusView:inWindow:usingBlock:] + 96
26  com.apple.AppKit              	0x00007fff22d2a6cb -[_NSViewBackingLayer display] + 382
27  com.apple.QuartzCore          	0x00007fff26be05b8 CA::Layer::display_if_needed(CA::Transaction*) + 874
28  com.apple.QuartzCore          	0x00007fff26d1a373 CA::Context::commit_transaction(CA::Transaction*, double, double*) + 517
29  com.apple.QuartzCore          	0x00007fff26bbdf91 CA::Transaction::commit() + 783
30  com.apple.AppKit              	0x00007fff22dd49cb __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
31  com.apple.AppKit              	0x00007fff2351ed0e ___NSRunLoopObserverCreateWithHandler_block_invoke + 41
32  com.apple.CoreFoundation      	0x00007fff20437d9d __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
33  com.apple.CoreFoundation      	0x00007fff20437c2d __CFRunLoopDoObservers + 549
[... 461 lines cut ...]
16  libsystem_pthread.dylib       	0x00007fff20340950 _pthread_start + 224
17  libsystem_pthread.dylib       	0x00007fff2033c47b thread_start + 15

Thread 31:: Java: Java2D Disposer
0   libsystem_kernel.dylib        	0x00007fff2030e8e2 __psynch_cvwait + 10
1   libsystem_pthread.dylib       	0x00007fff20340e6f _pthread_cond_wait + 1254
2   libjvm.dylib                  	0x000000010e4d38a6 os::PlatformEvent::park() + 192
3   libjvm.dylib                  	0x000000010e4cb40e ObjectMonitor::wait(long, bool, Thread*) + 748
4   libjvm.dylib                  	0x000000010e584c74 ObjectSynchronizer::wait(Handle, long, Thread*) + 202
5   libjvm.dylib                  	0x000000010e39ca23 JVM_MonitorWait + 156
6   ???                           	0x0000000111e869f4 0 + 4595411444
7   ???                           	0x0000000111e772bd 0 + 4595348157
8   ???                           	0x0000000111e77040 0 + 4595347520
9   ???                           	0x0000000111e77040 0 + 4595347520
10  ???                           	0x0000000111e77302 0 + 4595348226
11  ???                           	0x0000000111e6f7a7 0 + 4595316647
12  libjvm.dylib                  	0x000000010e33af8a JavaCalls::call_helper(JavaValue*, methodHandle*, JavaCallArguments*, Thread*) + 1710
13  libjvm.dylib                  	0x000000010e33b718 JavaCalls::call_virtual(JavaValue*, KlassHandle, Symbol*, Symbol*, JavaCallArguments*, Thread*) + 356
14  libjvm.dylib                  	0x000000010e33b8c4 JavaCalls::call_virtual(JavaValue*, Handle, KlassHandle, Symbol*, Symbol*, Thread*) + 74
15  libjvm.dylib                  	0x000000010e396039 thread_entry(JavaThread*, Thread*) + 124
16  libjvm.dylib                  	0x000000010e5b827b JavaThread::thread_main_inner() + 155
17  libjvm.dylib                  	0x000000010e5b9968 JavaThread::run() + 450
18  libjvm.dylib                  	0x000000010e4d7b0a java_start(Thread*) + 246
19  libsystem_pthread.dylib       	0x00007fff20340950 _pthread_start + 224
20  libsystem_pthread.dylib       	0x00007fff2033c47b thread_start + 15

MenKuch, 2020-11-15:
Additional Information: Tried another newly created user account (same result), deleting preferences (same result), redownloading (same result), downloading beta of Arduino IDE 1.19 (same result).

omid1979, 2020-11-16:
I have this similar problem , any idea ?

MenKuch, 2020-11-16:
I could fix the problem by updating Java on Big Sur. Head to the System Preferences and click on Java -> Update.

It would be nice if the Arduino IDE would fail in a more explanatory fashion :-)

sebromero, 2020-11-17:
The IDE comes with a bundled Java 8 runtime environment, so updating Java shouldn't be necessary 🤔

MenKuch, 2020-11-17:
Very strange indeed. I did not spent much time debugging this issue, but it was completely reproducible: Arduino IDE was crashing with the call stack I posted above and as soon as I updated the java version through macOS system preferences, it worked. Guessing, there seems to be a dependency to the java version installed on macOS despite Arduino having and embedded runtime.

DrSAR, 2020-11-18:
Confirmed here as well. When I hit 'Reopen' on the crash notice rather than 'Ignore', I get it to start on the second try. Seems like (fragile) work around but wth?!

Kwinnieprince, 2020-11-22:
I also have Big Sur 11.0.1. Arduino starts with the smash screen and then just stops without giving any error message.

ardu1, 2020-11-27:
Hello, 
I also have the same problem with the Arduino IDE.

I use the Arduino IDE since several month. Now I made the update on my 2017 MacBook to Big Sur. After all was ready I had the problem to upload any code to my esp8266. I find the solution here https://forum.arduino.cc/index.php?topic=702144.0 and it works now. But when I tried to use my esp32 I got again an error. But all I did to solve the problem does not work.

Then I took a second  MacBook (2017) with already installed Big Sur. I installed the Arduino IDE for the first time on this MacBook. After the download I copied the Folder in my applications folder and started the IDE. After a moment I saw the information, that there is a problem, should it start again? I tried it, then I got the same error and the possibility to send an error report to apple (it looks like the other here).

After I tried it 2-3 times the IDE was running. 

Then I tried to made the patches about the problems with the esp8266. But I was wondering, because in the folder "Arduini15" in the Library I didn't found the folder Java. I tried to install it again, but I had the same problem.

So it seams here is the problem?

ardu1, 2020-11-27:
Oh sorry, there is a mistake in relation to the Java folder. This folder does not exist in the Library, only in the program folder under applications is a Java folder. And in this folder I have to put the portable folder. Than I could see it in the Library under Arduino15.

adamelliot, 2020-12-03:
So I was running into the issues with Java. To fix this I removed all my installed Java versions. Every version works differently and it's all garbage. The one that is bundled with the IDE works.

https://explainjava.com/uninstall-java-macos/

claudeheintz, 2020-12-17:
It appears to me that Arduino does a search for JRE's in the /Library/Java/JavaVirtualMachines folder.  After much consternation and messing with java_home, I determined this by manually launching Arduino from the terminal and looking at the output.

I had a leftover java 1.8 folder in in the /Library/Java/JavaVirtualMachines folder.  I deleted just that folder and presto, Arduino worked.  I have remaining /Library/Java/JavaVirtualMachines/jdk-12.0.1.jdk and /Library/Java/JavaVirtualMachines/jdk-15.0.1.jdk folders that do not seem to bother Arduino, it was only the older 1.8.x version.

I deleted it.  So, I'm now not sure if it was  jre-1.8... or jdk-1.8...  But, getting rid of it solved my problem with Arduino launching.

Kwinnieprince, 2020-12-27:
@claudeheintz I also had a leftover in the /Library/Java/JavaVirtualMachines folder (jdk1.8.0_181.jdk) When I deleted that directory, Arduino just started as normal.
I did not have any other directories in that folder but I think it indeed has something to do with 1.8.x version

petertorelli, 2021-01-02:
Confirming @claudeheintz. Even though the IDE bundles Java, I removed an old 1.8.0_101 version from 2016 in that area (since the new Java version on macOS appears to be in the `Internet Plugins` directory) and now the Arduino IDE starts with no problem.

jemenake, 2021-01-02:
@claudeheintz Deleting the JDK's in /Library/Java/JavaVirtualMachines also fixed it for me with Arduino 1.8.13 on Big Sur.

per1234, 2021-02-06:
There are a couple other issues that seem to be about the same thing, which I'll mention here for the sake of keeping everything linked together:
- https://github.com/arduino/Arduino/issues/10936
- https://github.com/arduino/Arduino/issues/11055

And maybe even this one: https://github.com/arduino/Arduino/issues/11150

ValerioParisMitritsakis, 2021-03-08:
Deleting lingering old java versions did the trick for me as well.

imvenj, 2021-03-27:
I stuck with this problem for months and kept using 1.8.12. Today I finally see this thread and tried the method mentioned by @claudeheintz and it works like a charm. By deleting all Oracle 1.8.x JVMs, Arduino IDE 1.8.3 now starts normally. 

In case you DO need 1.8.x JVM, you can try OpenJDK 1.8 from [AdopeOpenJDK](https://adoptopenjdk.net). It won't crash Arduino IDE 1.8.3.

edant92, 2021-04-18:
The fix from @claudeheintz also worked for me.

Removing `jdk1.8.0_161.jdk` from `/Library/Java/JavaVirtualMachines` on macOS Big Sur 11.2.3 .

sgeffner22, 2021-07-26:
This fix does not work for me. I've uninstalled all version of Java on my Mac (MacOS version 11.5 (20G71)) and yet the Arduino editor still crashes on launch. I see the splash screen, and it successfully opens previously-opened windows, and then crashes. 

I've seen a lot of folks on this (and other) threads saying that removing java fixes the problem for them. I'll point out that Arduino is a hobby platform that is used by lots of kids and these kinds of workarounds are probably too hard for them to figure out. This issue really ought to be fixed and pushed in a broad release so it's not creating frustration for users, especially new users. (Plus, the workaround doesn't work for me, so I'd like to see the issue just fixed.)

jenschr, 2021-07-29:
I can confirm that this bug still happens to older Arduino versions on macOS Big Sur 11.2.3. Installing Arduino 1.8.15 seems to fix the issue, so I'd recommend others to do the same. There are [several](https://forum.arduino.cc/t/ide-1-8-10-crashes-often-on-mac-osx-10-13-6/621304) [threads](https://forum.arduino.cc/t/arduino-ide-keeps-crashing-mac/658158) showing people with this problem in the forums also.

jenschr, 2021-08-18:
This is a macOS issue and not really Arduino, but many users will encounter this so I would suggest detecting it in the Arduino installer to help this.

Typing this in Terminal:

`java -version`

... will show you the currently installed version. In my case, this returns:

```
java version "1.8.0_121"
Java(TM) SE Runtime Environment (build 1.8.0_121-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.121-b13, mixed mode)
```

To change your Java install to use OpenJDK and solve the issue, just follow these instructions: https://mkyong.com/java/how-to-install-java-on-mac-osx/

After installing and executing the commands that Homebrew suggests after the install, you should have a working install and the Java version command will return this:

```
openjdk version "1.8.0_302"
OpenJDK Runtime Environment (build 1.8.0_302-bre_2021_08_14_21_34-b00)
OpenJDK 64-Bit Server VM (build 25.302-b00, mixed mode)
```