arduino/Arduino #10970: Add RiversEngineering to Arduino Library Manager

yoshifujita, 2020-11-16:
https://github.com/RiversEngineering/RiversEngineering

per1234, 2020-11-17:
Hi @yoshifujita. Thanks so much for your interest in contributing to the Arduino Library Manager index!

Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The `name` value "Adafruit BMP3XX Library" used in your library (https://github.com/RiversEngineering/RiversEngineering/blob/2.0.0/library.properties#L1) has already been taken.

If you would still like your library added to the index, you'll need to change the `name` value to a unique name, then either update the `version` value in library.properties and make a new release/tag or else redo the existing tag.

After that is done, please comment here to let us know we can proceed. Thanks!

---
More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

RiversEngineering, 2020-11-17:
I fixed it. Forgot to save changes after copying the properties file from another library as a template.

cmaglie, 2020-11-18:
Done.

closed 2020-11-18 by cmaglie