arduino/Arduino #10975: [Library Manager] Please add the PlugPiBlack library

FlavianMF, 2020-11-18:
Plese add a [PlugPiBlack](https://github.com/picodebr/PlugPiBlack) to library manager

Thanks

per1234, 2020-11-18:
Hi @FlavianMF. Thanks for your interest in contributing to the Arduino Library Manager index!

The library.properties file in your library is not compliant with the Arduino Library specification, which is required before your library can be added to the Library Manager index:
https://arduino.github.io/arduino-cli/latest/library-specification/#libraryproperties-file-format

After bringing the library.properties file into compliance with the specification, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we'll know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

FlavianMF, 2020-11-18:
I made the necessary changes, could you check if everything is ok now?

Thanks

closed 2020-12-03 by cmaglie