arduino/Arduino #10976: [arduino.cc] Library reference page not updated after new version released

stechio, 2020-11-18:
Last week I released a new version ([3.0.0](https://github.com/stechio/arduino-ad-mux-lib/releases/tag/3.0.0)) of [Analog-Digital Multiplexers](https://github.com/stechio/arduino-ad-mux-lib) library.

The Library Manager indexer timely harvested it [as expected](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-is-the-library-list-generated) (the new tag appears in the [library feed](http://downloads.arduino.cc/libraries/library_index.json)) but, after ten days, _the library's reference page_ ([https://www.arduino.cc/reference/en/libraries/analog-digital-multiplexers/](https://www.arduino.cc/reference/en/libraries/analog-digital-multiplexers/)) _hasn't been updated yet_ (it still indicates 2.0.1 as latest release!).

Is there any issue with the site generator?

alranel, 2020-11-23:
Sorry @stechio, we're going to fix this very soon (thank you for the report)

stechio, 2020-12-01:
Today I surfed the page again and it looks OK: did you actually fix it or the generation job just moved on by itself? (in the latter case I would expect such random behavior to keep happening in the future...)

thanks

closed 2021-03-19 by stechio