arduino/Arduino #10979: New repository for FauxmoESP library

pvint, 2020-11-19:
Hello,

I've taken over maintenance of the FaumoESP library and it has been moved from the old repo on Bitbucket to a new one on Github.

Old repo: https://bitbucket.org/xoseperez/fauxmoesp/src/master/
New repo: https://github.com/vintlabs/fauxmoESP

All of the tagged releases have been migrated to the new repo, and I've updated the _library.properties_ file on both the old and new repos to reflect the changes.

Please let me know if there is anything else I must do to have the library manager updated to reflect the changes for the new repo.

Thanks!
Paul

For reference, here is the contents of the _library.properties_ file:
```
name=FauxmoESP
version=3.1.1
author=Xose Pérez <xose.perez@gmail.com>
maintainer=Paul Vint <paul@vintlabs.com>
sentence=Amazon Alexa support for ESP8266 and ESP32
paragraph=
category=Communication
url=https://github.com/vintlabs/fauxmoESP
architectures=esp8266,esp32
includes=fauxmoESP.h
```

cmaglie, 2020-12-03:
Done.

closed 2020-12-03 by cmaglie

pvint, 2020-12-03:
Excellent - thanks!

pvint, 2020-12-08:
The library browser in Arduino IDE is still referring to the old repo (bitbucket) rather than the new one (github). Is there something I have done wrong?

per1234, 2020-12-08:
@pvint do you mean the "More info" link? That is controlled by the library.properties `url` field, which still points to the bitbucket repositorry:
https://github.com/vintlabs/fauxmoESP/blob/3.1.1/library.properties#L8
```
url=https://bitbucket.org/xoseperez/fauxmoESP
```

pvint, 2020-12-08:
Well, that, and everything else too. ;)

I updated the library.properties in the master branch (https://github.com/vintlabs/fauxmoESP/blob/master/library.properties) as well as in the newest release. Do I need to change it in the previous releases?

per1234, 2020-12-08:
>as well as in the newest release

The library.properties `version` value wasn't updated before making that release so it was rejected by the indexer because the `version` value is used for versioning in the index and there was already a `3.1.1` in the index:
https://github.com/vintlabs/fauxmoESP/blob/3.1.2-RC1/library.properties#L2
```
version=3.1.1
```

The Arduino Library Manager index works exclusively from Git tags, so changes to the repository made after the time of the latest compliant tag have no effect on Library Manager.

pvint, 2020-12-08:
Ooops! I'll fix that.

Thanks for the help @per1234 !

pvint, 2020-12-09:
It's working perfectly now - many thanks!