arduino/Arduino #10982: IDE 1.8.13 on Linux does not start

gipsayan, 2020-11-20:
The same problem appeared on IDE 1.8.10

After a clean install the first time IDE starts correctly. From the second time on, it shows a quick splash screen and closes down immediately.
The reason is that during the first launch the IDE creates the ./arduino15 folder with the preferences.txt file
This file contains the entry "sketchbook.path=/home/user/Arduino" 
This entry is not accepted by the IDE.

Manual fix:
1. verify that a sketchbook subfolder is available in the Arduino folder
2. edit the "sketchbook.path" entry in the preferences.txt file so that it contains the full path:
"sketchbook.path=/home/user/Arduino/sketchbook" 

To IDE developers. It should be very easy to verify and solve this matter in the source code. Please fix it.

per1234, 2020-11-20:
I'll guess that you have the Arduino IDE installed at `/home/user/Arduino`.

If so, we already have a lot of issues about how this can be improved:

- #10766
- #10511
- #10572
- #7280
- #4995
- #4995
- #2719

So I'll go ahead and close this as a duplicate. If I am wrong about the cause, feel free to comment here with more information and I'll consider reopening the issue.

closed 2020-11-20 by per1234

gipsayan, 2020-11-20:
Fine with closing my issue. But given that it is a persistent problem affecting so many people, why it just isn't fixed? I do not know the IDE internals, but in other programming languages writing code for a file/folder verification would take only few minutes. A folder is missing? Make it! The entry in "preferences.txt" points to the wrong place? Take a default with the correct one.

matthijskooijman, 2020-11-20:
@gipsayan, there is the intention to fix it, but nobody has taken the time to do so (there's >9000 things to fix all the time...).

However, I'm wondering if this is really the same issue. #10766 suggests improving the error message you get in this situation, but this report suggest that it silently fails. Though in https://github.com/arduino/Arduino/issues/10511#issuecomment-667599079 @per1234 suggests that the error message is somehow broken and should be fixed.

Regardless, if anyone investigates this further and provides a PR to make sure you get an error message again (and maybe improve it as discussed in #10766), I'm happy to review that PR and see if we can get it merged.