arduino/Arduino #10986: Typo in hyperlink in language reference page for variables

bethouxa, 2020-11-22:
Both links to the scope documentation page found in [the "Variables" language reference page](https://www.arduino.cc/en/Reference/VariableDeclaration) redirect towards an URL that has a typo in it, which loads a blank page.

Both links are present in the two "Variable scope" sections, the links point towards a redirection page (https://www.arduino.cc/en/Reference/Scope) which itself redirects to [https://www.arduino.cc/reference/en/language/variables/variable-scope--qualifiers/scope/](https://www.arduino.cc/reference/en/language/variables/variable-scope--qualifiers/scope/), notice the two hyphens at the end of the address, whereas the actual page has a single hyphen.

Removing the hyphen manually fixes the issue and loads the intended page.

sam-2200, 2020-11-22:
@per1234 May I try to resolve this issue ,I am new to opensource

per1234, 2020-11-22:
Hi @sam-2200. Thanks so much for your interest in contributing to the Arduino project!

Unfortunately, the content of this tutorial is not hosted in a public repository, so there is no way for you to assist with it. One of the Arduino employees who have edit access to the tutorial's source content would need to fix the error.

If you would like to help to improve Arduino's documentation, the Arduino Language Reference is hosted here:
https://github.com/arduino/reference-en
so if you have any proposals for fixes or improvements, you're welcome to open pull requests or bug reports in that repository.

We also have initiatives to translate that content into multiple languages other than English. You can see the repositories for the translations here:
https://github.com/arduino?q=reference-