arduino/Arduino #10993: Switch/Case and If Statements Limit

peasoup49, 2020-11-23:
Please fix the limit of 'if' and/or 'switch/case' total in the compiler.  It strongly appears to be a memory allocation or limit that is set in the compiler and affects both 'if' and 'switch/case' statements alike.  Originally I had 155 switch/cases total throughout my program (each with 1 line command per case).  When I commented out 6 (bringing the total to 149) the program compiles and runs as expected.  Anything over that, the compiler will compile and upload, but the program will not run.  Any attempt to add more conditions per case, causes the same issue of the program not running if even more case statements are not commented out.  I went as far as re-writing approximately 25% of my program to replace all the 'switch/case' statements with 'if' statements.  Physical memory on the board is not a problem as this is where I'm currently at on my atMega2560:

Sketch uses 120778 bytes (47%) of program storage space. Maximum is 253952 bytes.
Global variables use 3034 bytes (37%) of dynamic memory, leaving 5158 bytes for local variables. Maximum is 8192 bytes.

matthijskooijman, 2020-11-23:
Interesting problem. Given we're using the standard gcc compiler, I would be somewhat surprised if there would be some maximum on the number of if statements (usually, problems like this end up being problems in the code someone has written), but we've seen weird compiler bugs, so it's not unthinkable. If it is a compiler problem, then I'd expect it's not so much the number of ifs that is limited somehow, but some other particularity of the code you've written that trips up the compiler.

To properly look into this, we'll need a complete example of code that ends up being broken, along with a clear description of what the code should do and actually does, so we can reproduce the problem and investigate this. Ideally, this is a minimal example that can show the problem (i.e. a sketch with just the ifs and and maybe some prints or so to show the problem). If that doesn't work (maybe a minimal example ends up being not broken, happens a lot), then you can try to strip down your original problem, removing parts one by one and leaving the ones that make the problem go away.

Also, please state the Arduino IDE version, exact board and core/platform version you are using (verbose compilation output should contain most of this info).

peasoup49, 2020-11-23:
I'm using the latest version of the IDE, 1.8.13 and the exact board is a Sundowner Mega 2560, their Ethernet Shield, and their RS232/485 Shield.
I assume it's this you're looking for??
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library SD at version 1.2.4 in folder: C:\Program Files (x86)\Arduino\libraries\SD 
Using library Ethernet at version 2.0.0 in folder: C:\Program Files (x86)\Arduino\libraries\Ethernet 
Using library TextFinder-master in folder: C:\Users\Mystery User\Documents\Arduino\libraries\TextFinder-master (legacy)

Here is the first switch/case with the last 6 cases commented out (valu) is an integer, (finder is the TextFinder Library):
`switch (valu)
              {
                case 1:
                  Code0x01(client);
                  break;

                case 2:
                  Code0x02(client);
                  break;

                case 3:
                  Code0x03(client);
                  break;

                case 4:
                  Code0x04(client);
                  break;

                case 5:
                  Code0x05(client);
                  break;

                case 6:
                  Code0x06(client);
                  break;

                case 7:
                  Code0x07(client);
                  break;

                case 8:
                  Code0x0A(client);
                  break;

                case 9:
                  Code0x0B(client);
                  break;

                case 10:
                  Code0x0F(client);
                  break;

                case 11:
                  Code0x10(client);
                  break;

                case 12:
                  Code0x11(client);
                  break;

                case 13:
                  Code0x12(client);
                  break;

                case 14:
                  Code0x13(client);
                  break;

                case 15:
                  Code0x14(client);
                  break;

[... 627 lines cut ...]
    case 152:
      client.print(F("<script>btn74.style.backgroundColor='#FFFF00'"
                     "btn74.style.color='#000000'</script>\n"));
      break;

    case 153:
      client.print(F("<script>btn75.style.backgroundColor='#FFFF00'"
                     "btn75.style.color='#000000'</script>\n"));
      break;

    case 154:
      client.print(F("<script>btn76.style.backgroundColor='#FFFF00'"
                     "btn76.style.color='#000000'</script>\n"));
      break;

    case 155:
      client.print(F("<script>btn77.style.backgroundColor='#FFFF00'"
                     "btn77.style.color='#000000'</script>\n"));
      break;

    default:
      break;

  }
}`

matthijskooijman, 2020-11-23:
Thanks for providing some more details. However, this shows the offending bit of code, which looks like it should work at first glance, but does not give me a *complete* example of something that is broken. With that I mean: A sketch that others can just copy-paste into the IDE and run on a board (e.g. an Arduino Mega2560, which is common enough for people to have it) to see the problem. Ideally this sketch can show the problem without additional hardware (i.e. without an Ethernet shield, which I and most people won't have...).

>  I assume it's this you're looking for??

There's also a bit at the start that says which core or platform (dunno which term is used there exactly) is used, with the version and directory that is helpful.

I've also edited your comment to fix the formatting a bit, you need to use triple backticks (`````````) instead of a single one for blocks of code.

matthijskooijman, 2020-11-26:
@peasoup49 sent me his sketch through private e-mail, since they didn't want to share it publicly yet. That's fine, but I want to keep discussion public, so I'm responding here.

@peasoup49, looking at the sketch you sent me, it still is your full sketch, not a minimal example like I asked for. In particular, it still relies on the Ethernet library and hardware, so I can't really test this locally to reproduce. You've also not clearly stated what the actual problem is ("It doesn't work" is too broad), so I can't really move forward here.

You also mention that you suspect that you might be running out of stack space. I can't be sure, but that would be a plausible explanation (when you run out of stack space, pretty much anything can happen, and removing random bits of code can help prevent the problem). With that, I now think that, as I originally suspected, this is just a bug in your code that causes some weirdness (possibly by running out of memory), rather than the compiler that actually produces incorrect results.

Since this issue tracker is for actual bugs in the IDE and compilation process, not for getting help with your own code, I'm going to close this issue. I also personally don't have time to help you find the issue, so you might want to seek assistance in other places, such as the forum or the IRC channel. Feel free to post followup comments here if you think there is a bug after all (we can always reopen) or maybe when you solve your issue (just for future reference).

closed 2020-11-26 by matthijskooijman