arduino/Arduino #11000: [Library Manager] Request to add the M5EPD library

Gitshaoxiang, 2020-11-26:
pls add the M5EPD library

[Github Link](https://github.com/m5stack/M5Core2/releases/tag/0.0.1)

thank you.

per1234, 2020-11-26:
Thanks for your interest in contributing to the Arduino Library Manager index @Gitshaoxiang!

Libraries containing .exe files are prohibited from being added to the Library Manager index. Your library contains a couple of them:
- https://github.com/m5stack/M5EPD/tree/main/tools/image2gray
- https://github.com/m5stack/M5EPD/tree/main/tools/ttf2bin

If you still want the library added to Library Manager, you'll need to remove these .exe files. You might consider keeping only the source code in the repository and providing the compiled binaries as release assets.

After removing the .exe files, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Gitshaoxiang, 2020-11-26:
now. we already delete the `.exe` file and release the tag.  thank you 

[Github Link](https://github.com/m5stack/M5EPD/releases/tag/0.1.0)

Gitshaoxiang, 2020-12-03:
hi.....  could you tell me the estimated time to add?  about this lib..... this product already for sale one week.    so we really need your help.

per1234, 2020-12-03:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall.

You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

closed 2020-12-03 by cmaglie