arduino/Arduino #11003: [Library Manager] Question

gmarty2000, 2020-11-27:
Hello, last year I asked you to add some libraries into the "Library Manager" (the request was accepted).
Now I would like to migrate them to another account, and still keep their URLs in the "Library Manager".
What kind of information do you need to update library's URLs?

per1234, 2020-11-27:
Hi @gmarty2000. Please just let us know the old and the new URL of each library's repository.

gmarty2000, 2020-11-27:
Thanks for the quick response! Here is the list of OLD and NEW library URLs.
|OLD|NEW|
|---|---|
|https://github.com/gmarty2000/arduino-BUZZER|https://github.com/gmarty2000-ARDUINO/arduino-BUZZER|
|https://github.com/gmarty2000/arduino-JOYSTICK|https://github.com/gmarty2000-ARDUINO/arduino-JOYSTICK|
|https://github.com/gmarty2000/arduino-ULTRASONIC_SENSOR|https://github.com/gmarty2000-ARDUINO/arduino-ULTRASONIC_SENSOR|
|https://github.com/gmarty2000/arduino-SOIL_HYGROMETER|https://github.com/gmarty2000-ARDUINO/arduino-SOIL_HYGROMETER|

cmaglie, 2020-12-03:
Done.

closed 2020-12-03 by cmaglie

gmarty2000, 2020-12-03:
Thank you so much