arduino/Arduino #11005: Website - Code Error - Opla IoT Kit - Getting Started Project - Get To Know The Carrier

alexandru-cohal, 2020-11-27:
The _Opla IoT Kit "Get To Know The Carrier"_ project is available on this page: 
[https://opla.arduino.cc/opla/module/carrier/lesson/get-to-know-the-carrier](https://opla.arduino.cc/opla/module/carrier/lesson/get-to-know-the-carrier) 

In the _"Gradually building code"_ section, it is stated in the explanations that _"These buttons are controlled by using the commands carrier.Button1.onTouchDown() and carrier.Button2.onTouchDown() inside two different conditionals"_ but in the example code are used the commands _carrier.Button0.onTouchDown()_ and _carrier.Button1.onTouchDown()_. The first command, _**carrier.Button0.onTouchDown()**_, generates the error _**" 'class MKRIoTCarrier' has no member named 'Button0'; did you mean 'Buttons'? "**_ when the code is compiled.

The example code and the explanations should be synchronized (i.e. both of them should use the _Button1_ and _Button2_ class _MKRIoTCarrier_ members and not _Button0_ which is not anymore a member since the [change](https://github.com/arduino-libraries/Arduino_MKRIoTCarrier/commit/0d6cc3a648efb53523f6f5b494b3d5e6559f89a9) made in the _Arduino_MKRIoTCarrier.h_ file on the 20th of November)

I am using the _Arduino Web Editor_ with the _Arduino_MKRIoTCarrier Library_ version _0.9.6_.

karlsoderby, 2020-11-30:
Thank you @alexandru-cohal, this issue has been resolved.

Fernand0177, 2020-12-17:
Hi @karlsoderby, the bug still there in [the page](https://opla.arduino.cc/opla/module/carrier/lesson/get-to-know-the-carrier).

karlsoderby, 2020-12-17:
Hi @Fernand0177 , sorry for the inconvenience, we have now updated the page.

Fernand0177, 2020-12-17:
> 
> 
> Hi @Fernand0177 , sorry for the inconvenience, we have now updated the page.

Thanks, now is working properly, only one more thing, the last piece of code, the "Full Code" section, only shows the last fragment of the code, and no the full code, how in the description says. @karlsoderby Thanks for your great work!

alexandru-cohal, 2020-12-18:
@karlsoderby This bug appeared again because in the new version of the _Arduino_MKRIoTCarrier Library_ ([version 0.9.7](https://github.com/arduino-libraries/Arduino_MKRIoTCarrier/releases/tag/v0.9.7) released on the 16th of December) the numbering of the buttons is between 0 and 4 and not between 1 and 5 like in the previous version (_version 0.9.6_) which I referred when I added this issue initially, right?
I remember that when this issue was closed for the first time, everything was fine but it was valid only for the _version 0.9.6_ which was the latest one at that time 😃

karlsoderby, 2021-01-04:
Hi @alexandru-cohal apologies for the late reply during the holidays. There was a short period of time where the code snippets were updated, but the library was not, which would explain why this issue was happening. I just tested the example in question and it worked. Please let us know if this issue remains!

per1234, 2021-01-04:
@karlsoderby how about the separate issue reported incidentally here by @Fernand0177:
>the last piece of code, the "Full Code" section, only shows the last fragment of the code, and no the full code, how in the description says

alexandru-cohal, 2021-01-04:
@karlsoderby Thank you for the explanation! I also tested the example now (with the version 0.9.7 of the _Arduino_MKRIoTCarrier_ library) and I checked the website and everything is fine now (regarding the original issue) 😊

karlsoderby, 2021-01-05:
@alexandru-cohal Happy to see everything working and thanks for your patience! :)

karlsoderby, 2021-01-05:
@per1234 This issue has been solved! Thanks for the reminder.

closed 2021-03-29 by per1234