arduino/Arduino #11006: random(21)-function only gives {0,7,21} on first call

ShantySuffisance, 2020-11-27:
Hi, 

```
int n;

void setup() {
  Serial.begin(9600);
  randomSeed(analogRead(A0));
  n = random(21);
  Serial.println(n); 
}
void loop() {
}
```
only outputs numbers from the set:  {0,7,21}. The same problem occurs at the first call of  random(21) in loop. Every other number than 21, that i tested, didn't reproduce the problem. 

incerely yours
shanty

matthijskooijman, 2020-11-27:
I can confirm this on an Arduino Uno, the above sketch outputs (on repeated resets):
```
0
0
14
7
0
14
7
```

Modifying the example to:

```
```
gives me
```
7
7
14
0
7
14
0
7
14
.... keeps repeating the same cycle
```

Seems like something is wrong with the way the seed works combined with passing a value to `random` somehow. I haven't investigated other than trying to reproduce this.

@ShantySuffisance, can you tell us what Arduino board you've been testing on to confirm we're seeing the same issue?

ShantySuffisance, 2020-11-29:
@matthijskooijman i've tested on two Nanos, a Mega and a simulated Uno (tinkercad) 

thank you for your modified example, it really supports your hypothesis! 

i modified your modified example to test combinations of seed and random_max: 

```
int n;
int seed = 0;
int random_max = 0;

void setup() {
  Serial.begin(9600);
  for (seed = 0; seed < 255; seed++) {
    //Serial.print("new seed: "); Serial.println(seed); 
    for (random_max = 0 ; random_max < 255 ; random_max += 7) {
      randomSeed(seed);
      n = random(random_max);
      if ( n % 7) { // if n is NOT a multiple of 7:
        Serial.println("NOT multiples of 7 for: ");
        Serial.print("Seed: \t"); Serial.print(seed);
        Serial.print(" \t random-max: \t"); Serial.println(random_max);
      }
    }
  }
}
void loop() {
}
```

_Conclusion:_ only for seed = 0 and random_max >= 21 there are results which aren't divisible by seven…

matthijskooijman, 2020-11-30:
Thanks for confirming, that means this is a problem in the AVR core, so I'm moving this issue. I don't have time to look into this in detail now, but the relevant code is here, if you want to have a look: https://github.com/arduino/ArduinoCore-avr/blob/2f67c916f6ab6193c404eebe22efe901e0f9542d/cores/arduino/WMath.cpp

From a quick glance at the code, I *think* the Arduino side of things looks good, so maybe the libc `random()` function does something unexpected here (likely not wrong, just unexpected and probably documented).