arduino/Arduino #11009: Website - Code Error - Opla IoT Kit - Getting Started Project - Get To Know The Carrier - Acceleration Reading

alexandru-cohal, 2020-11-28:
The _Opla IoT Kit "Get To Know The Carrier"_ project is available on this page:
[https://opla.arduino.cc/opla/module/carrier/lesson/get-to-know-the-carrier](https://opla.arduino.cc/opla/module/carrier/lesson/get-to-know-the-carrier)

In the _"Core programming terms"_ section, _"Loop"_ subsection, there is a code snippet with commands for reading the sensors available on the _MKR IoT Carrier_. For reading the acceleration provided by the IMU sensor, the following command is provided
```
carrier.IMUmodule.readAccelerometer(accelerometer_x, accelerometer_y, accelerometer_z); //read accelerometer (x, y z)
```
which generates the following error 
```
error: 'class LSM6DS3Class' has no member named 'readAccelerometer'; did you mean 'readAcceleration'?
```
As suggested in the error and also explained in the documentation of the [LSM6DS3 library](https://www.arduino.cc/en/Reference/Arduino_LSM6DS3) of the IMU sensor, the _readAccelerometer_ method should be replaced with the _readAcceleration_ method.

I am using the _Arduino Web Editor_ with the _Arduino_MKRIoTCarrier Library_ version _0.9.6_.

karlsoderby, 2020-11-30:
Resolved. Thank you @per1234

karlsoderby, 2020-11-30:
Thank you @alexandru-cohal for reporting. This issue will be fixed shortly.

per1234, 2020-12-24:
@karlsoderby did you ever get a chance to fix this error? I see it is still there on the tutorial.

karlsoderby, 2021-01-04:
@per1234 just updated the platform, replacing **readAccelerometer** with **readAcceleration**. Thanks for the reminder, it had avoided previous updates somehow!

closed 2021-01-04 by per1234