arduino/Arduino #11017: mrm-* libraries for MRMS modular CAN Bus robots

PribaNosati, 2020-11-29:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
Can add the following libraries the Library Manager, in https://github.com/PribaNosati/MRMS:
mrm-8x8a in https://github.com/PribaNosati/MRMS/mrm-8x8a,
mrm-bldc2x50 in https://github.com/PribaNosati/MRMS/mrm-bldc2x50,
mrm-bldc4x2.5 in https://github.com/PribaNosati/MRMS/mrm-bldc4x2.5,
mrm-board in https://github.com/PribaNosati/MRMS/mrm-board,
mrm-can-bus in https://github.com/PribaNosati/MRMS/mrm-can-bus,
mrm-col in https://github.com/PribaNosati/MRMS/mrm-col,
mrm-common in https://github.com/PribaNosati/MRMS/mrm-common,
mrm-imu in https://github.com/PribaNosati/MRMS/mrm-imu,
mrm-ir-finder-can in https://github.com/PribaNosati/MRMS/mrm-ir-finder-can,
mrm-ir-finder2 in https://github.com/PribaNosati/MRMS/mrm-ir-finder2,
mrm-ir-finder3 in https://github.com/PribaNosati/MRMS/mrm-ir-finder3,
mrm-lid-can-b in https://github.com/PribaNosati/MRMS/mrm-lid-can-b,
mrm-lid-can-b2 in https://github.com/PribaNosati/MRMS/mrm-lid-can-b2,
mrm-lid1 in https://github.com/PribaNosati/MRMS/mrm-lid1,
mrm-lid2 in https://github.com/PribaNosati/MRMS/mrm-lid2,
mrm-mot2x50 in https://github.com/PribaNosati/MRMS/mrm-mot2x50,
mrm-mot4x10 in https://github.com/PribaNosati/MRMS/mrm-mot4x10,
mrm-mot4x3.6can in https://github.com/PribaNosati/MRMS/mrm-mot4x3.6can,
mrm-node in https://github.com/PribaNosati/MRMS/mrm-node,
mrm-pid in https://github.com/PribaNosati/MRMS/mrm-pid,
mrm-ref-can in https://github.com/PribaNosati/MRMS/mrm-ref-can,
mrm-robot in https://github.com/PribaNosati/MRMS/mrm-robot,
mrm-servo in https://github.com/PribaNosati/MRMS/mrm-servo,
mrm-switch in https://github.com/PribaNosati/MRMS/mrm-switch,
mrm-therm-b-can in https://github.com/PribaNosati/MRMS/mrm-therm-b-can.
The libraries support modular robots and CAN Bus peripherals based on Espressif ESP32 MCU.

per1234, 2020-11-29:
Thanks for your interest in contributing to the Arduino Library Manager index @PribaNosati.

In order to be added to the index, each library must be in the root of a dedicated repository. So it will not be possible for these libraries in subfolders of a single repository to be added. You would need to move each library to a separate repository.

There are some additional requirements listed here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ
for example, the library must not contain a `.development` file, which I noticed in a few of your libraries.

So if you still want your libraries added, please bring them into compliance with the requirements listed in the FAQ and then comment here with links to the repositories. If you no longer wish to have your libraries in the Library Manager index, you can feel free to close this issue.

PribaNosati, 2020-11-29:
Thanks. I will correct the issues.

closed 2020-11-29 by PribaNosati