arduino/Arduino #11018: Libraries for MRMS Arduino modular CAN Bus robots

PribaNosati, 2020-11-29:
Can add the following libraries the Library Manager:
mrm-8x8a in https://github.com/PribaNosati/mrm-8x8a,
mrm-bldc2x50 in https://github.com/PribaNosati/mrm-bldc2x50,
mrm-bldc4x2.5 in https://github.com/PribaNosati/mrm-bldc4x2.5,
mrm-board in https://github.com/PribaNosati/mrm-board,
mrm-can-bus in https://github.com/PribaNosati/mrm-can-bus,
mrm-col in https://github.com/PribaNosati/mrm-col,
mrm-common in https://github.com/PribaNosati/mrm-common,
mrm-imu in https://github.com/PribaNosati/mrm-imu,
mrm-ir-finder-can in https://github.com/PribaNosati/mrm-ir-finder-can,
mrm-ir-finder2 in https://github.com/PribaNosati/mrm-ir-finder2,
mrm-ir-finder3 in https://github.com/PribaNosati/mrm-ir-finder3,
mrm-lid-can-b in https://github.com/PribaNosati/mrm-lid-can-b,
mrm-lid-can-b2 in https://github.com/PribaNosati/mrm-lid-can-b2,
mrm-lid1 in https://github.com/PribaNosati/mrm-lid1,
mrm-lid2 in https://github.com/PribaNosati/mrm-lid2,
mrm-mot2x50 in https://github.com/PribaNosati/mrm-mot2x50,
mrm-mot4x10 in https://github.com/PribaNosati/mrm-mot4x10,
mrm-mot4x3.6can in https://github.com/PribaNosati/mrm-mot4x3.6can,
mrm-node in https://github.com/PribaNosati/mrm-node,
mrm-pid in https://github.com/PribaNosati/mrm-pid,
mrm-ref-can in https://github.com/PribaNosati/mrm-ref-can,
mrm-robot in https://github.com/PribaNosati/mrm-robot,
mrm-servo in https://github.com/PribaNosati/mrm-servo,
mrm-switch in https://github.com/PribaNosati/mrm-switch,
mrm-therm-b-can in https://github.com/PribaNosati/mrm-therm-b-can.
The libraries support modular robots and CAN Bus peripherals based on Espressif ESP32 MCU.

per1234, 2020-11-29:
>mrm-bldc4x2.5 in https://github.com/PribaNosati/mrm-bldc4x2.5,

I get a 404 error for this URL. Does that repository exist? If so, is the visibility of the repository set to "public"?

---
>mrm-col in https://github.com/PribaNosati/mrm-col,

It looks like this repository is currently in a compliant state. However, the Library Manager indexer works from Git tags, and at the time of the latest tag in the repository, it was not in a compliant state:
https://github.com/PribaNosati/mrm-col/tree/0.0.0
Please either create a new tag or else redo the existing one.

---
>mrm-common in https://github.com/PribaNosati/mrm-common,

It looks like this repository is currently in a compliant state. However, the Library Manager indexer works from Git tags, and at the time of the latest tag in the repository, it was not in a compliant state:
https://github.com/PribaNosati/mrm-common/tree/0.0.0
Please either create a new tag or else redo the existing one.

---
>mrm-lid2 in https://github.com/PribaNosati/mrm-lid2,

It looks like this repository is currently in a compliant state. However, the Library Manager indexer works from Git tags, and at the time of the latest tag in the repository, it was not in a compliant state:
https://github.com/PribaNosati/mrm-lid2/tree/0.0.0
Please either create a new tag or else redo the existing one.

---
>mrm-therm-b-can in https://github.com/PribaNosati/mrm-therm-b-can.

It looks like this repository is currently in a compliant state. However, the Library Manager indexer works from Git tags, and at the time of the latest tag in the repository, it was not in a compliant state:
https://github.com/PribaNosati/mrm-therm-b-can/tree/0.0.0
Please either create a new tag or else redo the existing one.

---
Once you have resolved these issues, please comment here so we'll know we can proceed with adding your libraries to the index.

PribaNosati, 2020-11-29:
Thank You for a fast response. I set mrm-bldc4x2.5 visibility to public and added tags for the rest of the mentioned libraries. I hope that will solve the issues.

cmaglie, 2020-12-03:
Done.

closed 2020-12-03 by cmaglie