arduino/Arduino #11021: MLX90614 I2C communication on Arduino Nano Every

omersa, 2020-12-01:
I have bought 6pack of Arduino Nano Every and built a splecial shield plate, so I can connect thermal sensor MLX90614, which communicate over I2C (A4,A5). The sensor was well tested on few Nano devices and on all it's OK. But with Arduino Nano Every it does not work. The device can be found on port I2C (search address), but no communication is going through. No data can be read or write. Tried many libraries and the problem persists with all.

As this device is has different processor then classic Nano, it seems that libraries for megaAVR boards are not OK. Also I am not alone with this problem. I think it persist on all Wire communication with Every.

alranel, 2020-12-01:
Hi @omersa, could you provide some more details about your setup?
Which libraries are you using? Did you add pullups? Are you using a "naked" thermal sensor or does it have a breakout?

omersa, 2020-12-01:
Sensor is in a 5V package with SDA and SCL out and is working well on all other Nano boards. The connection of SDA is A4 port and SCL A5 port, as normally. When you query the device on address 0x5A it confirms the connection, but no data is transmitted or received. Normally I am using Adafruit_MLX90614 library, but is also not working with any other wire type library.