arduino/Arduino #11024: IDE starts but does not shows

felixNilo, 2020-12-02:
Hello everyone.
I'm having problems with IDE. I don't know why the IDE starts but doesn't shows.
The output from arduino_debug.exe is quite long and does not shows a possible problem.

[debugOutput.txt](https://github.com/arduino/Arduino/files/5629467/debugOutput.txt)

I have been stucked in this issue the last two days, and I already tried to remove the file C:\Users\Username\AppData\Local\Arduino15\package_index.json and remove the folder C:\Users\Username\AppData\Local\Arduino15\cache

Helps pleaaaase.

SarveshOO7, 2020-12-30:
Hi
Does the log change after removing the file and/or the folder? Is this the log/output after you have deleted the file/folder or before? Could you share the log/output after deleting this file/folder?
Thankyou

felixNilo, 2020-12-30:
> Hi
> Does the log change after removing the file and/or the folder? Is this the log/output after you have deleted the file/folder or before? Could you share the log/output after deleting this file/folder?
> Thankyou

I have solved the problem by removing all the folders related to arduino, then, I installed the IDE directly from the Microsfot Store and there was no problem with that.

Thank you BTW for your response.

Cheers and have a happy new year.

closed 2020-12-30 by felixNilo