arduino/Arduino #11027: [Library Manager] Please add ESP-Mail-Client to Library Manager

mobizt, 2020-12-03:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

I would to add this ESP-Mail-Client library to the library manager
https://github.com/mobizt/ESP-Mail-Client

Due to compatibilities issue and the name changes to support ESP8266 device,
Can you remove this library https://github.com/mobizt/ESP32-Mail-Client from the library manager?

Thank you for your support.

mobizt, 2020-12-12:
Why the process takes 9 days without any feedback?

per1234, 2020-12-12:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. So the delay before the library is added is dependent on what point in the cycle your request happened to fall. You will receive a notification from GitHub when the issue is closed. After that, there may be a delay of some hours for the library's releases to be picked up by the indexer and for the updates to the [Library Manager index](http://downloads.arduino.cc/libraries/library_index.json) to propagate through the CDN cache. So we'll get to it soon, please be patient.

closed 2020-12-23 by cmaglie