arduino/Arduino #11029: Option to notify the user about library/board updates

drws, 2020-12-03:
It would be a helpful feature if Arduino IDE provided a way to enable some form of notification about available library and board updates ("Updatable" category in Library Manager/Boards Manager). Currently user has to manually check both, which are even in different submenus.

per1234, 2020-12-03:
Hi @drws. Thanks for taking the time to request a feature!

You'll be happy to learn that Arduino IDE already has this feature!
[image]

Along with the check for available IDE version update, it is enabled by the **File > Preferences > Check for updates on startup** preference, so make sure to turn that on. Note that the update check is not done immediately on startup, so don't be too impatient for the notification. I'm not sure what the trigger is, but if there are updateable boards platforms or libraries, you'll be sure to get the notification about it eventually.

closed 2020-12-03 by per1234

drws, 2020-12-07:
Thank you for the thorough explanation. I did have the _Check for updates on startup_ option disabled since I thought it only applies to the IDE version check itself (and I have a system package manager handling that).

I think it's best I don't open another feature request, but please also consider separating the _Check for updates on startup_ into three separate options for IDE, board and library updates. There are use cases where one wouldn't want to update everything and currently has to disable all checks to not be annoyed by a repeating notification. Separating makes it also more descriptive as to what the option means anyway (my case above for example).