arduino/Arduino #11030: Library Manager includes all files in repository instead of just those tagged in my GitHub release?

HomeSpan, 2020-12-04:
My HomeSpan library was just included in the Arduino Library Manager - thank you. However, the library manager appears to be downloading all files in the repository, not just the files that are part of source.zip when I created the specific GitHub tag.  Am I labeling something incorrectly? I want to make sure that only the files that have been bundled in the tagged release are downloaded by the library manager.

per1234, 2020-12-04:
I'll add a bit more information to make sure the situation is clear to the person investigating this issue:

I see that you have given some paths in your repository the [`export-ignore` attribute](https://git-scm.com/docs/gitattributes#_creating_an_archive):
https://github.com/HomeSpan/HomeSpan/blob/1.0.1/.gitattributes

Meaning the archive will have a folder structure like:
```
├── examples
├── library.properties
├── LICENSE
├── Other Examples
└── src
```
But the archive provided by the Library Manager index contains the "ignored" paths, resulting in a structure like:
```
├── docs
├── documents
├── examples
├── library.properties
├── LICENSE
├── Other Examples
└── src
```

HomeSpan, 2020-12-04:
Yes, that's exactly the problem.  Thanks.

per1234, 2020-12-08:
I discovered a previous issue about this: https://github.com/arduino/Arduino/issues/8478
It's best for us to have only one open issue for each topic in order to consolidate all the information in a single place, so I'm going to close this one in favor of the original. You can subscribe to that issue to get updates of any progress.

closed 2020-12-08 by per1234