arduino/Arduino #11033: Cannot compile 1.8.6 from source, possible issue: Java version number

kleinesfilmroellchen, 2020-12-04:
For the Arduino IDE support in Visual Studio Code, it seems like one needs to use a version <=1.8.6, as the README over there suggests. There are no binaries for this version, so I decided to compile from source. I followed the official instructions and have all prerequisites properly installed. Everything works fine until I encounter the following error in the `ant dist`:

```
C:\PATH_TO_ARDUINO_GIT\build\build.xml:159: The following error occurred while executing this line:
C:\PATH_TO_ARDUINO_GIT\build\build.xml:136: The following error occurred while executing this line:
C:\PATH_TO_ARDUINO_GIT\build\build.xml:1161: java.lang.ExceptionInInitializerError
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:145)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:181)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:169)
        at net.sf.launch4j.config.ConfigPersister.<init>(ConfigPersister.java:73)
        at net.sf.launch4j.config.ConfigPersister.<clinit>(ConfigPersister.java:66)
        at net.sf.launch4j.ant.Launch4jTask.execute(Launch4jTask.java:82)
        at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
        at jdk.internal.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:564)
        at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:99)
        at org.apache.tools.ant.Task.perform(Task.java:350)
        at org.apache.tools.ant.Target.execute(Target.java:449)
        at org.apache.tools.ant.Target.performTasks(Target.java:470)
        at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1401)
        at org.apache.tools.ant.helper.SingleCheckExecutor.executeTargets(SingleCheckExecutor.java:36)
        at org.apache.tools.ant.Project.executeTargets(Project.java:1264)
        at org.apache.tools.ant.taskdefs.Ant.execute(Ant.java:437)
        at org.apache.tools.ant.taskdefs.CallTarget.execute(CallTarget.java:106)
        at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
        at jdk.internal.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:564)
        at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:99)
        at org.apache.tools.ant.Task.perform(Task.java:350)
        at org.apache.tools.ant.Target.execute(Target.java:449)
        at org.apache.tools.ant.Target.performTasks(Target.java:470)
        at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1401)
        at org.apache.tools.ant.helper.SingleCheckExecutor.executeTargets(SingleCheckExecutor.java:36)
        at org.apache.tools.ant.Project.executeTargets(Project.java:1264)
        at org.apache.tools.ant.taskdefs.Ant.execute(Ant.java:437)
        at org.apache.tools.ant.taskdefs.CallTarget.execute(CallTarget.java:106)
        at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
        at jdk.internal.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:564)
        at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:99)
        at org.apache.tools.ant.Task.perform(Task.java:350)
        at org.apache.tools.ant.Target.execute(Target.java:449)
        at org.apache.tools.ant.Target.performTasks(Target.java:470)
        at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1401)
        at org.apache.tools.ant.Project.executeTarget(Project.java:1374)
        at org.apache.tools.ant.helper.DefaultExecutor.executeTargets(DefaultExecutor.java:41)
        at org.apache.tools.ant.Project.executeTargets(Project.java:1264)
        at org.apache.tools.ant.Main.runBuild(Main.java:827)
        at org.apache.tools.ant.Main.startAnt(Main.java:223)
        at org.apache.tools.ant.launch.Launcher.run(Launcher.java:284)
        at org.apache.tools.ant.launch.Launcher.main(Launcher.java:101)
Caused by: java.lang.StringIndexOutOfBoundsException: begin 0, end 3, length 2
        at java.base/java.lang.String.checkBoundsBeginEnd(String.java:3734)
        at java.base/java.lang.String.substring(String.java:1903)
        at com.thoughtworks.xstream.core.JVM.getMajorJavaVersion(JVM.java:25)
        at com.thoughtworks.xstream.core.JVM.<clinit>(JVM.java:12)
        ... 48 more

Total time: 18 seconds
```

Guessing by the last exception and its stack trace, the code is checking the version string, which has changed its anatomy after the move from 1.8 naming to 9., 10., ... naming in Java 9. I reported a similar issue on Processing a couple of years back, where the solution was "don't use Java 9+", which is ridiculous. (Just before you start suggesting that.) So I'm even suspecting that the issue is similar code, because AFAIK the Arduino IDE is forked from Processing.

Just so you know, I checked and the install instructions don't suggest any particular version of Java, although it seems like they were tested with OpenJDK 1.8. My PATH and JAVA_HOME point to OpenJDK 15. I also tried specifying some other target versions below 1.8.6 and they fail identically.

Any help would be appreciated. Or if there are still mirrors to pre-built 1.8 versions before 1.8.7 (Windows), that would be nice as well.

kleinesfilmroellchen, 2020-12-04:
Update: Found the old Processing bug report of mine: https://github.com/processing/processing/issues/5610. Just looked into it and it seems like Processing still doesn't support Java 9+.

davidair, 2021-03-26:
I ran into exactly the same issue today when building the Arduino IDE on Windows 10 using Cygwin and the latest version of OpenJDK I installed via choco.

As a workaround, I grabbed OpenJDK8 32-bit via https://www.openlogic.com/openjdk-downloads and was able to successfully build and run the IDE.