arduino/Arduino #11041: [Library Manager] Please add TinyInstruments to Library Manager

montoyamoraga, 2020-12-07:
https://github.com/montoyamoraga/TinyTrainable

thanks!

per1234, 2020-12-07:
Hi @montoyamoraga. Thanks for your interest in contributing to the Arduino Library Manager index.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

montoyamoraga, 2020-12-07:
hi @per1234, i think i just did, is that correct?
thank you :)

cmaglie, 2020-12-23:
Done. sorry for the long delay!

closed 2020-12-23 by cmaglie