arduino/Arduino #11043: Please add https://github.com/Zanduino/RotaryEncoder - a library to control a 3-color LED rotary encoder

SV-Zanshin, 2020-12-08:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

per1234, 2020-12-08:
Thanks for your contributions to the Arduino Library Manager index @SV-Zanshin!

Each library added to the Library Manager index must have a unique library.properties `name` value (regardless of case). The "RotaryEncoder" `name` value [used in the latest release of your library](https://github.com/Zanduino/RotaryEncoder/blob/v1.0.5/library.properties#L1) has already been taken:
https://github.com/mathertel/RotaryEncoder

I see that the `name` value has been changed to the unique "LEDRotaryEncoder" since that time, but there has been no new release and the Library Manager indexer works from releases.

You'll need to update the `version` value in library.properties and make a new release/tag before your library can be added.

After that is done, please comment here to let us know we can proceed. Thanks!

Note that the only requirement is for the library.properties `name` value to be unique. There are no such restrictions on the repository name or header file names of your library.

---
More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

SV-Zanshin, 2020-12-09:
Thanks for finding my errors. I've updated the library name to "RotaryEncoder_Zanduino" in library.properties and created a new, updated version and changed the version number in the library.properties to reflect that.

per1234, 2020-12-09:
The latest tag/release is from 2028 and still has the taken "RotaryEncoder" name:
https://github.com/Zanduino/RotaryEncoder/tags
https://github.com/Zanduino/RotaryEncoder/blob/v1.0.5/library.properties#L1

Did you maybe forget to push the tag?

SV-Zanshin, 2020-12-09:
I did neglect to push the tag. I've corrected that now.

cmaglie, 2020-12-23:
Done.

closed 2020-12-23 by cmaglie