arduino/Arduino #11044: Please add a library to access the Bosch BME280 - https://github.com/Zanduino/BME280

SV-Zanshin, 2020-12-08:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

per1234, 2020-12-08:
I see this library is already in the Library Manager index. But the owner name was changed from `SV-Zanshin` to `Zanduino`. So I guess this is really a request to update the URL rather than to add a new library..
```json
{
  "name": "BME280_Zanshin",
  "version": "1.0.4",
  "author": "https://github.com/SV-Zanshin",
  "maintainer": "https://github.com/SV-Zanshin",
  "sentence": "Access the Bosch BME280 temperature, humidity and pressure sensor",
  "paragraph": "Access the Bosch BME280 temperature, humidity and pressure sensor",
  "website": "https://github.com/SV-Zanshin/BME280",
  "category": "Device Control",
  "architectures": [
    "*"
  ],
  "types": [
    "Contributed"
  ],
  "repository": "https://github.com/SV-Zanshin/BME280.git",
  "url": "http://downloads.arduino.cc/libraries/github.com/SV-Zanshin/BME280_Zanshin-1.0.4.zip",
  "archiveFileName": "BME280_Zanshin-1.0.4.zip",
  "size": 746245,
  "checksum": "SHA-256:bf1a19ff0dad2a32ff5029111b1048dfa0b911ae7c570e6213046a2aab6476f4"
}
```

SV-Zanshin, 2020-12-09:
I wasn't sure if the library were already in the library manager, as I went into the Arduino 1.8.13 IDE and selected the library manager and performed a search for "BME280" and didn't see the library. But I repeated that this morning and see that I missed the library after all, please excuse my error.

Since I moved the library from "SV-Zanshin" to "Zanduino" I see that all other references moved with it, but am unsure of whether I need to update the library URL in the library manager or not.  

I plan on migrating several libraries from the old location to the new one in the near future (it is taking a while, as I am also changing from using TRAVIS-CI to GitHub actions at the same time). Do I need to request the owner name change using this issue system, or will the system redirect automatically?

Thanks,

-Arnd.

per1234, 2020-12-09:
>I see that all other references moved with it, but am unsure of whether I need to update the library URL in the library manager or not.

GitHub does automatically set up a redirect when the library is transferred, but it's probably best for us to update the URL in the Library Manager index to point to the real repository so we'll be sure it will continue to work.

>Do I need to request the owner name change using this issue system, or will the system redirect automatically?

Please also submit issues here when you migrate the other libraries.

SV-Zanshin, 2020-12-09:
Thanks - I'll do that.

cmaglie, 2020-12-23:
Done.

closed 2020-12-23 by cmaglie