arduino/Arduino #11051: Please add our library to interface with SFM10R4 modems

paulporto, 2020-12-09:
Please add our library to control and communicate with SFM10R4 modems
https://github.com/TECA-IOT/Tinyfox
thank you

per1234, 2020-12-09:
Thanks for your interest in contributing to the Arduino Library Manager index @paulporto.

Before we can add your library, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

paulporto, 2020-12-09:
im sorry, my partner was slow. its released now
thanks

paulporto, 2020-12-21:
its everything ok now? we are exited for this to be part of the official repo, if there is any problem please write to us. thank you.

cmaglie, 2020-12-23:
Done.

closed 2020-12-23 by cmaglie