arduino/Arduino #11052: please add our library for our devkit

paulporto, 2020-12-09:
please add our library for our new devkit, it has a sigfox modem and a arduino leonardo compatible hardware (32u4)
https://github.com/TECA-IOT/Ufox/tree/v1.0.0
many thanks

per1234, 2020-12-10:
Hi @paulporto. The library.properties file is misnamed "library.properties..properties":
https://github.com/TECA-IOT/Ufox/blob/v1.0.0/library.properties..properties
In order for the library to be added to the Library Manager index, the library.properties file must have that exact filename.

After renaming the library.properties file, you will need to create a new tag/release or redo the existing one.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://arduino.github.io/arduino-cli/latest/library-specification/#library-metadata

paulporto, 2020-12-10:
fixed. thank you for your patience.

per1234, 2020-12-10:
OK, the repository looks good now, but at the time of the tag it was not compliant:
https://github.com/TECA-IOT/Ufox/tree/v1.0.0
Please either create a new tag/release or else redo the existing one, then comment here to let us know it's done.

paulporto, 2020-12-10:
im sorry. we make a new release
https://github.com/TECA-IOT/Ufox/releases/tag/v1.0.1

paulporto, 2020-12-21:
its everything ok now? we are exited for this to be part of the official repo, if there is any problem please write to us. thank you.

per1234, 2020-12-21:
@paulporto yes, everything looks fine to me now. Your library will be added after cmaglie gets back from vacation. This issue will be closed at that time so you'll get a notification.

cmaglie, 2020-12-23:
Done.

closed 2020-12-23 by cmaglie