arduino/Arduino #11055: An installed Oracle Java crashes Arduino IDE 1.8.13 on macos

morungos, 2020-12-10:
When attempting to start the IDE on a Mac with Java installed, it crashes out after showing the splash screen. It's not entirely predictable, because I've been using the IDE with reasonable success until a week or so ago. However, there is enough that this seems to be a bug. It appears to be searching for Java versions and accidentally finding the browser plugin one rather than an actual functioning Java. Erasing Oracle Java from the system (as per: https://stackoverflow.com/questions/19039752/removing-java-8-jdk-from-mac) does resolve the problem, and the IDE then starts successfully.

OS: Macos 11.0.1
Arduino IDE: 1.8.13
Oracle Java versions installed: jdk1.8.0_111 and jdk1.7.0_80.jdk 
(Note: switching off browser plugins did not solve the problem).

The problem shows up in the command line when starting the IDE directly. With Oracle Java installed, key lines:

```
2020-12-10 17:01:47.362 Arduino[1815:26249] JVMArguments=(
)
2020-12-10 17:01:47.362 Arduino[1815:26249] JVMClasspath=(null)
2020-12-10 17:01:47.362 Arduino[1815:26249] JVMDefaultOptions={
}
2020-12-10 17:01:47.362 Arduino[1815:26249] -> Bundle path: /Applications/Arduino.app
2020-12-10 17:01:47.362 Arduino[1815:26249] -> Working Directory: '/Users/stuart'
2020-12-10 17:01:47.362 Arduino[1815:26249] -> JVM Runtime path: /Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre
2020-12-10 17:01:47.363 Arduino[1815:26249] Searching for a Java 8 virtual machine
2020-12-10 17:01:47.363 Arduino[1815:26249] Search for java VM in '/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home'
2020-12-10 17:01:47.457 Arduino[1815:26249] -> JVM Runtime path updated to: /Library/Internet Plug-Ins/JavaAppletPlugin.plugin
2020-12-10 17:01:47.457 Arduino[1815:26249] -> Java Runtime Dylib Path: '/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/lib/jli/libjli.dylib'
2020-12-10 17:01:47.458 Arduino[1815:26249] Command line passed to application argc=21:
2020-12-10 17:01:47.458 Arduino[1815:26249] Arg 0: '/Applications/Arduino.app/Contents/MacOS/Arduino'
...
```

There is more, but it's verbose and doesn't seem particularly relevant. However, the `JVM Runtime path updated to: /Library/Internet Plug-Ins/JavaAppletPlugin.plugin` does seem to be a useful sign of an erroneous startup step.

The startup then fails with:

```
Set log4j store directory /Users/stuart/Library/Arduino15
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007fff203820a9, pid=1815, tid=0x0000000000000307
#
# JRE version: Java(TM) SE Runtime Environment (8.0_201-b09) (build 1.8.0_201-b09)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.201-b09 mixed mode bsd-amd64 compressed oops)
# Problematic frame:
# C  [libsystem_platform.dylib+0x10a9]  _platform_memmove$VARIANT$Haswell+0x29
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /Users/stuart/hs_err_pid1815.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

I doubt if the more detailed log is useful, but I'll add if requested. However, one line does jump out -- showing a mixup between `java.ext.dirs` and `java.library.path` that looks like a harbinger of doom. 

```
jvm_args: -Djava.library.path=/Applications/Arduino.app/Contents/MacOS -DLibraryDirectory=/Users/stuart/Library -DDocumentsDirectory=/Users/stuart/Documents -DApplicationSupportDirectory=/Users/stuart/Library/Application Support -DCachesDirectory=/Users/stuart/Library/Caches -DSandboxEnabled=true -Dapple.awt.application.name=Arduino -Dcom.apple.macos.use-file-dialog-packages=true -Dcom.apple.smallTabs=true -DAPP_DIR=/Applications/Arduino.app/Contents/Java -Djava.ext.dirs=/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/lib/ext/:/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/jre/lib/ext/ -Djava.net.preferIPv4Stack=true -Dcom.apple.mrj.application.apple.menu.about.name=Arduino -Dfile.encoding=UTF-8 -Xms128M -Xmx512M 
```

After removing Oracle Java (I do still have an OpenJDK install):

```
2020-12-10 17:07:28.341 Arduino[1393:8717] JVMArguments=(
)
2020-12-10 17:07:28.341 Arduino[1393:8717] JVMClasspath=(null)
2020-12-10 17:07:28.341 Arduino[1393:8717] JVMDefaultOptions={
}
2020-12-10 17:07:28.341 Arduino[1393:8717] -> Bundle path: /Applications/Arduino.app
2020-12-10 17:07:28.341 Arduino[1393:8717] -> Working Directory: '/Users/stuart'
2020-12-10 17:07:28.341 Arduino[1393:8717] -> JVM Runtime path: /Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre
2020-12-10 17:07:28.341 Arduino[1393:8717] Searching for a Java 8 virtual machine
2020-12-10 17:07:28.341 Arduino[1393:8717] Search for java VM in '/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home'
2020-12-10 17:07:28.342 Arduino[1393:8717]   KO - error: 'launch path not accessible'
2020-12-10 17:07:28.357 Arduino[1393:8717] Search for java VM in '/Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home'
2020-12-10 17:07:28.463 Arduino[1393:8717]   KO - invalid 'java -version' output
2020-12-10 17:07:28.463 Arduino[1393:8717] -> Java Runtime Dylib Path: '/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/lib/jli/libjli.dylib'
2020-12-10 17:07:28.465 Arduino[1393:8717] Command line passed to application argc=21:
2020-12-10 17:07:28.465 Arduino[1393:8717] Arg 0: '/Applications/Arduino.app/Contents/MacOS/Arduino'
```

Despite these log lines, the IDE seems to start and run fine.  

This seems to be fairly easy to work around by simply removing Oracle Java entirely (unless you really need Oracle Java, and let's be honest, who does?) but the user experience is not especially helpful (a vanilla crash) so it would be great if the IDE could avoid this issue in future.

Note that there is some similarity to #9828, and there are mentions of issues relating to the Java bundled inside Arduino on the in the forums, but this seems to be a distinct issue, as origins seem to be more in Oracle Java and its Mac paths than anything else.

per1234, 2020-12-10:
Thanks so much for your report and excellent analysis of the problem @morungos! If I understand correctly, this is the same as https://github.com/arduino/Arduino/issues/10964. Is that correct?

morungos, 2020-12-10:
Great question, I'd somehow managed to miss that one. They're certainly very closely related, and yes: I think this is probably a duplicate of that. Certainly, I can use the IDE on Big Sur -- so long as I removed installed Oracle Java directories. I also have an OpenJDK/adoptopenjdk version in `/Library/Java/JavaVirtualMachines` and that doesn't seem to cause the problem. And I see that in #10964, the IDE is similarly modifying the JVM path to a non-app Java (in that case: `jdk1.8.0_181.jdk`), so the common theme of the `JVM Runtime path updated to` log line does seem to be implicated. 

I do have a feeling that, like in [the comment by DrSAR](https://github.com/arduino/Arduino/issues/10964#issuecomment-729854169), I once got accidentally get through the crash to a working state, even before and on Big Sur. Similarly I have vague memories of something not dissimilar happening in Catalina. In any event, it does seem to be the search for (and acceptance of) an unrelated Java 8 virtual machine from the system that triggers the havoc, but I'd be slightly surprised if this was specific to Big Sur. 

There are reports of Java changes in Big Sur (see: https://stackoverflow.com/questions/64917779/wrong-java-home-after-upgrade-to-macos-big-sur-v11-0-1/64917842) but there's no evidence that `java_home`/`JAVA_HOME` is ever used. That does break Java IDEs like Eclipse, but... my `$JAVA_HOME` is now empty, and yet the Arduino IDE is working, and `/usr/libexec/java_home` does find my OpenJDK one. 

I did a little digging and the prime suspect seems to me to be Appbundler. Some of the log statements match old open source forks, but not all. So I'm wondering whether it's being either too smart or too stupid to correctly handle the Mac case of a specific, provided, bundled JRE -- and this might explain why Big Sur does seem to be more exposed. But my Java skills are a little rusty.

sbliven, 2021-02-08:
I'm not sure which of these duplicate issues are definitive, but this has a descriptive title so I'll comment here.

There are a few steps that should be taken to fix this:
1. Remove  /Library/Internet Plug-Ins/JavaAppletPlugin.plugin and /Library/Java/JavaVirtualMachines/jdk1.8.0_192.jdk from the search path (they are broken and outdated)
2. Allow configuring the location, e.g. through JAVA_HOME, plist.info, etc
3. Update to use a more recent jre internally