arduino/Arduino #11059: Please add Arduino_GFX library

moononournation, 2020-12-11:
Library link: https://github.com/moononournation/Arduino_GFX.git

per1234, 2020-12-12:
Thanks for your interest in contributing to the Arduino Library Manager index @moononournation

Arduino has a policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

Your current `name` value ("Arduino_GFX") is in violation of this rule:
https://github.com/moononournation/Arduino_GFX/blob/master/library.properties#L1
```
name=Arduino_GFX
```

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

After that, create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag).

Once you have done this, please comment here so we'll know we can proceed.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

moononournation, 2020-12-12:
Ok, revised and tagged v1.0.1

cmaglie, 2020-12-23:
Done.

moononournation, 2021-04-11:
I found both v1.0.8 and v1.0.9 cannot install correctly, the last installable version is v1.0.7.
Is it something conflicted in Arduino_GFX new version?

per1234, 2021-04-11:
I see that files were added after the v1.0.7 tag where the path differs only in case:
https://github.com/moononournation/Arduino_GFX/commit/cd730849f0aa2de9d8c9db77214845468e561167

For example:

- https://github.com/moononournation/Arduino_GFX/blob/v1.0.8/examples/ImgViewerAnimatedGIF/Data/ezgif.com-optimize.gif
- https://github.com/moononournation/Arduino_GFX/blob/v1.0.8/examples/ImgViewerAnimatedGIF/data/ezgif.com-optimize.gif

(**D**ata vs. **data**)

While this is perfectly fine on a case sensitive file system like Linux, the two files can not coexist on case insensitive file systems like Windows or macOS. I suspect this is the cause of the silent failure of the Library Manager installation of the 1.0.8 and 1.0.9 versions of the library.

moononournation, 2021-04-12:
tested v1.1.0 ok after revised the folders. Please help to remove corrupted v1.0.8 and v1.0.9.

per1234, 2021-04-12:
I'm glad to hear it! I saw you had given removing those redundant files a try (https://github.com/moononournation/Arduino_GFX/commit/6bb45a7ca176fcab4ae2caf6d94a8175bdfc8a6e).

I notice you have already removed the problematic tags, in accordance with [the FAQ's instructions](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#sorry-i-did-something-wrong-how-can-i-change-or-unpublish-an-already-published-library), so now I think it will just be a matter of waiting for Cristian to work some magic on the next pass through the Library Manager requests.

cmaglie, 2021-04-26:
Done.

closed 2021-04-26 by cmaglie