arduino/Arduino #11068: IDE prompts to update boards that is already up to date.

tikonen, 2020-12-14:
I get constant pop-ups to update 'boards or libraries'. The board IDE proposes to update is "Arduino nRF528x Boards (Mbed OS)". However the latest version 1.1.6 seems to be already installed.  
[image]

There is also a strange "1.3.0" folder in my mbed installation that seems to contain also installation. Where that could have came from? It may be related to the problem.

`C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\hardware\mbed>dir
 Volume in drive C has no label.
 Volume Serial Number is BC4F-7093

 Directory of C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\hardware\mbed

11/09/2020  09:14 PM    <DIR>          .
11/09/2020  09:14 PM    <DIR>          ..
08/24/2020  09:08 AM    <DIR>          1.1.6
11/09/2020  09:17 PM    <DIR>          1.3.0
               0 File(s)              0 bytes
               4 Dir(s)  25,598,328,832 bytes free

C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\hardware\mbed>dir 1.1.6
 Volume in drive C has no label.
 Volume Serial Number is BC4F-7093

 Directory of C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.6

08/24/2020  09:08 AM    <DIR>          .
08/24/2020  09:08 AM    <DIR>          ..
10/28/2019  04:58 PM               225 .mbedignore
07/28/2020  10:08 AM             1,923 boards.txt
07/19/2019  04:08 PM    <DIR>          bootloaders
07/19/2019  04:08 PM    <DIR>          cores
09/11/2019  05:13 PM    <DIR>          drivers
07/31/2019  10:55 AM    <DIR>          libraries
07/28/2020  10:08 AM             8,347 mbed-os-to-arduino
07/29/2019  03:56 PM                92 mbed_lib.json
07/28/2020  10:08 AM    <DIR>          patches
07/28/2020  02:02 PM             6,344 platform.txt
09/11/2019  05:12 PM               600 post_install.bat
07/19/2019  04:08 PM               178 programmers.txt
07/31/2019  03:53 PM             1,393 README.md
07/19/2019  04:08 PM    <DIR>          variants
               8 File(s)         19,102 bytes
               8 Dir(s)  25,598,328,832 bytes free

C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\hardware\mbed>dir 1.3.0
 Volume in drive C has no label.
 Volume Serial Number is BC4F-7093

 Directory of C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0

11/09/2020  09:17 PM    <DIR>          .
11/09/2020  09:17 PM    <DIR>          ..
10/12/2020  12:56 PM               190 .mbedignore
10/12/2020  12:56 PM             1,286 bfg_nano33ble.sh
10/12/2020  12:56 PM             5,821 boards.txt
10/12/2020  12:56 PM    <DIR>          bootloaders
10/12/2020  12:56 PM    <DIR>          cores
10/12/2020  12:56 PM    <DIR>          drivers
10/12/2020  12:56 PM    <DIR>          libraries
10/12/2020  12:56 PM             8,900 mbed-os-to-arduino
10/12/2020  12:56 PM                92 mbed_lib.json
10/12/2020  12:56 PM    <DIR>          patches
10/12/2020  01:02 PM             8,726 platform.txt
10/12/2020  12:56 PM               783 post_install.bat
10/12/2020  12:56 PM               341 programmers.txt
10/12/2020  12:56 PM             2,289 README.md
10/12/2020  12:56 PM    <DIR>          tools
10/12/2020  12:56 PM    <DIR>          variants
               9 File(s)         28,428 bytes
               9 Dir(s)  25,598,394,368 bytes free

C:\Users\Teemu Ikonen\AppData\Local\Arduino15\packages\arduino\hardware\mbed>`

facchinm, 2020-12-14:
Hi @tikonen ,
it looks like you encountered a know bug in the board manager filter that matches the title and not the actaul core. You can safely remove 1.1.6 folder manually and start using 1.3.0 (that contains support for both Nano33BLE and PortentaH7). After removing the folder please restart the IDE; the new menus should be called `Arduino MbedOS Boards (nRF52840 / STM32H747)`

facchinm, 2020-12-14:
This issue will be fixed as `arduino-cli` integration is complete

closed 2020-12-14 by facchinm

tikonen, 2020-12-14:
That did the trick, thank you!
[image]