arduino/Arduino #11069: Arduino IDE menu options scroll off the screen in Linux Mint

iWa1rus, 2020-12-14:
Hi there, Not sure if this is specific to Mint or if it affects Ubuntu/Debian versions too (I don't have a machine running Debian to test with presently), but I have a large number of example snippets available to use, and the menu is disappearing off the bottom of my screen when I go to select an option. There are menu items that I need to access that are off the screen. I can use my cursor buttons to navigate to these menu items and their sub-menu items show at the bottom of the screen, so I know they are there, but the lower bounding box of the scroll window is not taking the active screen's size into account. 

This does not happen on another machine set up identically to this one with the same version of Mint and the same IDE version, but for one exception: One reason this could be occurring (I have not tested my theory) is I have multiple displays that are unevenly tiled on this machine, but on the other machine they are evenly positioned meaning there is a lower boundary for the entire screen area, but it is out of the display's boundary. You'll see what I mean on the images below. 

Dragging the window over to the laptop's screen on the right does not fix the issue, the menu still bounds correctly at the top of the screen, but rolls off the bottom without giving access to scroll the menu.

OS: Linux Mint 19.3 Cinnamon (x64)
Arduino IDE Version: 1.8.13 (I am unsure of the bitness though seem to recall downloading the 64 bit version on this machine)

[image]

Showing tiling of my screens:
[image]

iWa1rus, 2020-12-15:
Follow up note - I've checked other UI menus in the IDE that extend beyond the bottom of the screen, anything that exceeds the vertical resolution is going over without bounding and adding scroll buttons, it's not limited to the example snippet menu.

SarveshOO7, 2020-12-26:
I was unable to reproduce this issue in Ubuntu 20.04.1 LTS. I haven't tried to reproduce it in Linux Mint yet. But it seems to me that this issue is only present in Linux Mint.
OS: Ubuntu 20.04.1 LTS
Version: 1.8.14

SarveshOO7, 2020-12-27:
I confirm that I was unable to reproduce this bug on Linux Mint 19.3 Cinnamon (x64) and Arduino IDE Version: 1.8.13. Hence this might be an issue specific to users with unevenly tiled external display screens.

[image]

fpistm, 2021-04-09:
Hi,
I have a similar issue.
Arduino IDE 1.8.13 on Windows 10. Examples menu has the correct behavior but not in Tools menu.
The new STM32duino core release will provide more boards support but now list are longer than screen height, and impossible to scroll down the list like in the example menu.
[image]
[image]

/cc @matthijskooijman FYI

matthijskooijman, 2021-04-09:
@fpistm, I think your issue is a different one. You're using (might say abusing :-p) board options to do the actual board selection, but nobody ever considered that option menus would ever be so big that they need to be scrolled, so no scroller is created for those menus. It's quite a trivial fix, though. If you create a separate issue for this to not pollute this issue, I'll whip up a quick PR to fix it.

fpistm, 2021-04-09:
Thanks @matthijskooijman 
Done with #11416