arduino/Arduino #11075: [Library Manager] Please add Pulse Express library

protocentralashwin, 2020-12-16:
Please add this library to library manager: https://github.com/Protocentral/protocentral-pulse-express

Thanks in advance.

Ashwin

per1234, 2020-12-16:
Hi @protocentralashwin. The `name` value defined in your library.properties contains a prohibited character (`,`).
https://github.com/Protocentral/protocentral-pulse-express/blob/1.0.0/library.properties#L1
```
name=ProtoCentral Pulse Express SpO2, HR and BPT sensor
```

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

Then update the `version` value in library.properties and create a new [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag), or else redo the existing release/tag.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

protocentralashwin, 2020-12-16:
Thank you @per1234 

Sorry about that. I have made the correction and made a new release tag 1.0.1

cmaglie, 2020-12-23:
Done.

closed 2020-12-23 by cmaglie