arduino/Arduino #11078: install.sh Totally borked up my system

yolofy, 2020-12-17:
https://github.com/arduino/Arduino/blob/f4e8a91f101773c47481ae5444e7b4e32ddda0a4/build/linux/dist/install.sh#L1

This installation script totally messed up my system, all icons in gnome-shell where missing, launching apps was impossible. Had to completely reinstall gnome to fix it. I won't try to reproduce. I don't know what happend but here are some clues:

First the extracted location contained a space in the directory name:
/home/user1/Downloads/arduino-1.8.13-linux64 (1)/arduino-1.8.13

Strange messages found in syslog
Dec 17 00:00:26 COMP gnome-shell[3957]: Could not load a pixbuf from icon theme.#012This may indicate that pixbuf loaders or the mime database could not be found.
Dec 17 00:00:27 COMP gnome-shell[3957]: Failed to load image: Unrecognized image file format
Dec 17 00:00:27 COMP gnome-shell[3957]: Bogus presentation time 0 travelled back in time, using current time.
Dec 17 00:00:27 COMP gnome-shell[3957]: Failed to load file:///var/lib/AccountsService/icons/user1: Unrecognized image file format
Dec 17 00:00:31 COMP gnome-shell[3957]: Removing a network device that was not added
Dec 17 00:00:49 COMP gnome-shell[3957]: Can't update stage views actor StLabel is on because it needs an allocation.
Dec 17 00:00:49 COMP gnome-shell[3957]: Can't update stage views actor ClutterText is on because it needs an allocation.
Dec 17 00:01:01 COMP gnome-shell[9148]: Telepathy is not available, chat integration will be disabled.
Dec 17 00:01:02 COMP gnome-shell[9148]: Failed to load background 'file:///home/user1/.local/share/backgrounds/mybackground.jpg': Unrecognized image file format
Dec 17 00:01:02 COMP gnome-shell[9148]: Could not load a pixbuf from /org/gnome/shell/theme/icons/scalable/status/message-indicator-symbolic.svg.#012This may indicate that pixbuf loaders or the mime database could not be found.
Dec 17 00:01:02 COMP gnome-shell[9148]: st_widget_get_theme_node called on the widget [0x5625e0c8c2e0 Gjs_ubuntu-dock_ubuntu_com_docking_DashToDock.bottom shrink running-dots extended:insensitive "dashtodockContainer"] which is not in the stage.
Dec 17 00:01:02 COMP gnome-shell[9148]: Spurious clutter_actor_allocate called for actor 0x5625e0c8c2e0/dashtodockContainer which isn't a descendent of the stage!
Dec 17 00:01:03 COMP gnome-shell[3957]: gnome-shell: Fatal IO error 0 (Success) on X server :0.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor Gjs_ui_appDisplay_AppIcon is on because it needs an allocation.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor StWidget is on because it needs an allocation.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor Gjs_ui_iconGrid_BaseIcon is on because it needs an allocation.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor StBoxLayout is on because it needs an allocation.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor StBin is on because it needs an allocation.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor StIcon is on because it needs an allocation.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor ClutterActor is on because it needs an allocation.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor StLabel is on because it needs an allocation.
Dec 17 00:01:07 COMP gnome-shell[9148]: Can't update stage views actor ClutterText is on because it needs an allocation.
Dec 17 00:02:08 COMP gnome-shell[9148]: ../clutter/clutter/clutter-offscreen-effect.c:239: Unable to create an Offscreen buffer
Dec 17 00:02:55 COMP gnome-shell[9148]: message repeated 947 times: [ ../clutter/clutter/clutter-offscreen-effect.c:239: Unable to create an Offscreen buffer]
Dec 17 00:15:20 COMP gnome-shell[9148]: gnome-shell: Fatal IO error 11 (Resource temporarily unavailable) on X server :1.
Dec 17 01:40:31 COMP /usr/libexec/gdm-x-session[15844]: dbus-daemon[15844]: [session uid=123 pid=15844] Reloaded configuration
Dec 17 01:44:20 COMP gnome-shell[8983]: Window manager warning: Window 0x340002e sets an MWM hint indicating it isn't resizable, but sets min size 1 x 1 and max size 2147483647 x 2147483647; this doesn't make much sense.
Dec 17 01:47:50 COMP gnome-shell[8983]: Window manager warning: MetaSyncRing: Sync object is not ready -- were events handled properly?
Dec 17 01:47:50 COMP gnome-shell[8983]: message repeated 2 times: [ Window manager warning: MetaSyncRing: Sync object is not ready -- were events handled properly?]
Dec 17 01:47:50 COMP gnome-shell[8983]: Window manager warning: MetaSyncRing: Too many reboots -- disabling

matthijskooijman, 2020-12-17:
Hm, that sounds bad, sorry for the trouble that you went through.

> First the extracted location contained a space in the directory name:

This will probably cause problems. The install script seems to *mostly* handle spaces properly by quoting, but I see two lines where quoting must be added:

https://github.com/arduino/Arduino/blob/f4e8a91f101773c47481ae5444e7b4e32ddda0a4/build/linux/dist/install.sh#L13
https://github.com/arduino/Arduino/blob/f4e8a91f101773c47481ae5444e7b4e32ddda0a4/build/linux/dist/install.sh#L116

However, I would think that this should only cause various commands to fail with an error message, rather than actually do something harmful, so I'm not sure this explains the problems you are seeing.

As for the messages in syslog, that doesn't really help me understand what was going on. Also, gnome/gtk errors or assertion failures in the logs are not really uncommon, even on working systems, so it would ble interesting to see if there are errors that had never occured before (but I suppose those log messages are no longer retrievable).

>  Had to completely reinstall gnome to fix it

What does this mean exactly? Reinstall OS packages? Or remove some `~/.gnome-something` directory or so?

Did you run the install script as root, or under your regular user?

yolofy, 2020-12-17:
>However, I would think that this should only cause various commands to fail with an error message, rather than actually do something harmful, so I'm not sure this explains the problems you are seeing.

It might be xdg-icon-resource, xdg-desktop-menu or xdg-mime not handling this properly, in that case there should be some kind of pre-check before executing.

> As for the messages in syslog, that doesn't really help me understand what was going on. Also, gnome/gtk errors or assertion failures in the logs are not really uncommon, even on working systems, so [...]

Most suspicous messages are:
`Could not load a pixbuf from icon theme.#012This may indicate that pixbuf loaders or the mime database could not be found.`
and
`Failed to load file:///var/lib/AccountsService/icons/user1: Unrecognized image file format`

>What does this mean exactly? Reinstall OS packages? Or remove some ~/.gnome-something directory or so?

I followed this guide https://technastic.com/install-stock-gnome-shell-on-ubuntu/