arduino/Arduino #11083: [Library Manager] Request to add the Simple_HCSR04 library.

mobergmann, 2020-12-18:
Hello, I would like it if you could add the [Simple_HCSR04](https://github.com/mobergmann/Simple_HCSR04) library to the library manager.

Repo URL: https://github.com/mobergmann/Simple_HCSR04

Thank you very much

per1234, 2020-12-18:
Hi @mobergmann. Thanks for your interest in contributing to the Arduino Library Manager index!

In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://arduino.github.io/arduino-cli/library-specification/#libraryproperties-file-format

After adding the library.properties file, you will need to create a new tag/release or redo the existing one. Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

mobergmann, 2020-12-18:
I already added these files as far as I know

mobergmann, 2020-12-18:
Wait, there is a file missing! i need to check this out

mobergmann, 2020-12-18:
I forgot to push the changed, sorry. 
Now it's up to date @per1234

per1234, 2020-12-18:
OK, now please either create a new tag/release or else redo the existing one. The Library Manager indexer works from git tags and, even though your repository is currently compliant with the requirements, it was not at the time the 1.0 tag was created:
https://github.com/mobergmann/Simple_HCSR04/tree/1.0

mobergmann, 2020-12-18:
Thanks for your fast response, I recreated the relase

mobergmann, 2020-12-18:
Do I need to do something?

per1234, 2020-12-18:
No. Everything looks fine now. 

Your library will be available for installation via Library Manager once it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub at that time.

mobergmann, 2020-12-18:
Ahhh, thank you

cmaglie, 2020-12-23:
Done.

closed 2020-12-23 by cmaglie