arduino/Arduino #11089: GTK2 EOL

NicoHood, 2020-12-20:
GTK2 has reached its end of life. Correct me if I am wrong, but the arduino ide still uses GTK2? That should be fixed now, almost every other application is GTK3 ready now.

Also see:
https://archlinux.org/todo/gtk-2-eol/
https://blog.gtk.org/2020/12/16/gtk-4-0/

CReimer, 2021-01-12:
I ran ```pacman -Rdd gtk2``` and your arduino package still works fine. So no GTK2 dependency?

NicoHood, 2021-09-28:
Some [recently opened another issue about arduino and gtk2](https://bugs.archlinux.org/task/72265) for archlinux. He pointed out, that there were some gtk2 related comments in the changelog:
https://github.com/arduino/Arduino/blob/c33fb9fde3cdd6c5d6629fa2a672ef19c110902a/build/shared/revisions.txt#L176

So is gtk2 support now fully optional? Do we still need gtk3 then or even 4 or maybe none at all?

xyproto, 2022-08-26:
@NicoHood It doesn't look like much is happening on the upstream side with regards to upgrading Arduino from GTK+2. How do you feel about changing the `arduino` package to makedep on `gtk2-compat` and depend on `gtk3`?

NicoHood, 2022-08-27:
Does this work? What is the difference between gtk2-compat and gtk3?

xyproto, 2022-08-27:
@NicoHood gtk2-compat uses gtk3 but offers gtk2-like configuration files and header files. It does not work for every project, but it appears to work for arduino.