arduino/Arduino #11095: "Add .zip-library" getting "Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException"

MarphXL, 2020-12-22:
I try it with a new version with installer and the latest stable portable version and the nightly build and the latest beta.
I uninstalled everything and deleted the folder "C:\Users\Marph.000\AppData\Local\Arduino15".
But I still get this error, when I try to "add .zip-library" (Menu/Sketch/include library/add .zip library).

Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at sun.awt.shell.Win32ShellFolder2.getFileSystemPath(Win32ShellFolder2.java:571)
	at sun.awt.shell.Win32ShellFolder2.access$400(Win32ShellFolder2.java:72)
	at sun.awt.shell.Win32ShellFolder2$2.call(Win32ShellFolder2.java:298)
	at sun.awt.shell.Win32ShellFolder2$2.call(Win32ShellFolder2.java:296)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at sun.awt.shell.Win32ShellFolderManager2$ComInvoker$3.run(Win32ShellFolderManager2.java:556)
	at java.lang.Thread.run(Thread.java:748)

MarphXL, 2020-12-22:
Next try: uninstalled alle versions of arduino, running clean-registry and installed the Microsoft-store-version. rebooted and The error still exists.

per1234, 2020-12-22:
Thanks for the report @MarphXL. Does the error occur as soon as you open the menu item, or after you have selected the .zip file and clicked the "Open" button?

MarphXL, 2020-12-23:
The error occures when I click on the menuitem "add .zip library". No file dialog or something else happened.

SarveshOO7, 2020-12-27:
I believe the error is being caused by one of two reasons

- usage of Windows XP or older and Java 6 or lower. This issue was fixed in Java 8. This would imply that this is not an error caused by the arduino and that install a newer OS with Java 8 preinstalled/ installed externally.

-  by the fact that in the .arduino15 folder in the preferences.txt there might be an issue with the value assigned to sketchbook.path.

`      File libFolder = sourceFile;
      if (FileUtils.isSubDirectory(new File(PreferencesData.get("sketchbook.path")), libFolder)) {
        activeEditor.statusError(tr("A subfolder of your sketchbook is not a valid library"));
        return;
      }

      if (FileUtils.isSubDirectory(libFolder, new File(PreferencesData.get("sketchbook.path")))) {
        activeEditor.statusError(tr("You can't import a folder that contains your sketchbook"));
        return;
      }`

MarphXL, 2020-12-27:
Thx for your answer. I will check the "preferences.txt" and the sketchbook.path
I'm running a well updated Win10 with newest Java-Version.

SarveshOO7, 2020-12-30:
was your issue solved? @MarphXL  If yes then we can close this issue. @per1234