arduino/Arduino #11097: "KeyUsage does not allow digital signatures" when Board Manager attempts downloading "package_index.json".

bowdi, 2020-12-22:
Arduino IDE: 1.8.13
Windows Version: Windows 10  Enterprise, 10.0.18363 Build 18363

I was getting a "unable to find valid certification path to requested target" error as described in #8474 as I'm behind a company proxy but after applying [this](https://github.com/arduino/Arduino/issues/8474#issuecomment-494798961) fix I now get a "KeyUsage does not allow digital signatures" error as below:
```
Preparing boards...
2020-12-22T15:18:51.552Z INFO c.a.c.p.ContributionInstaller:305 [main] Start download and signature check of=[https://downloads.arduino.cc/packages/package_index.json]
Downloading platforms index... 
2020-12-22T15:18:51.556Z INFO c.a.u.n.FileDownloaderCache:92 [main] Cache folder C:\Users\P611654\AppData\Local\Arduino15\cache
2020-12-22T15:18:51.588Z INFO c.a.u.n.FileDownloaderCache:149 [main] Get file cached is expire true, exist false, info FileCached{eTag='null', lastETag='null', remoteURL='https://downloads.arduino.cc/packages/package_index.json', localPath='C:\Users\P611654\AppData\Local\Arduino15\cache\downloads.arduino.cc\packages\package_index.json', md5='null', createdAt='2020-12-22T15:18:51.569', cacheControl=null} 
2020-12-22T15:18:52.890Z INFO c.a.u.n.HttpConnectionManager:153 [cc.arduino.packages.discoverers.serial.SerialDiscovery] Connect to https://builder.arduino.cc/builder/v1/boards/0x0403/0x6001, method=GET, request id=604266058A7743FD
2020-12-22T15:18:52.890Z INFO c.a.u.n.HttpConnectionManager:153 [main] Connect to https://downloads.arduino.cc/packages/package_index.json, method=HEAD, request id=C15803D0E1D24A3F
2020-12-22T15:18:53.105Z ERROR c.a.u.n.FileDownloader:199 [main] The request stop
javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: KeyUsage does not allow digital signatures
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1946) ~[?:1.8.0_191]
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:316) ~[?:1.8.0_191]
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:310) ~[?:1.8.0_191]
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1639) ~[?:1.8.0_191]
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:223) ~[?:1.8.0_191]
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1037) ~[?:1.8.0_191]
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:965) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1064) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1367) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1395) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1379) ~[?:1.8.0_191]
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559) ~[?:1.8.0_191]
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185) ~[?:1.8.0_191]
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:162) ~[?:1.8.0_191]
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:155) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:106) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloaderCache.updateCacheInfo(FileDownloaderCache.java:184) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloaderCache.getFileCached(FileDownloaderCache.java:153) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:167) [arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129) [arduino-core.jar:?]
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:147) [arduino-core.jar:?]
	at cc.arduino.contributions.DownloadableContributionsDownloader.downloadIndexAndSignature(DownloadableContributionsDownloader.java:165) [arduino-core.jar:?]
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:306) [arduino-core.jar:?]
	at processing.app.Base.<init>(Base.java:318) [pde.jar:?]
	at processing.app.Base.main(Base.java:150) [pde.jar:?]
Caused by: sun.security.validator.ValidatorException: KeyUsage does not allow digital signatures
	at sun.security.validator.EndEntityChecker.checkTLSServer(EndEntityChecker.java:271) ~[?:1.8.0_191]
	at sun.security.validator.EndEntityChecker.check(EndEntityChecker.java:143) ~[?:1.8.0_191]
	at sun.security.validator.Validator.validate(Validator.java:274) ~[?:1.8.0_191]
	at sun.security.ssl.X509TrustManagerImpl.validate(X509TrustManagerImpl.java:324) ~[?:1.8.0_191]
	at sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:229) ~[?:1.8.0_191]
	at sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:124) ~[?:1.8.0_191]
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1621) ~[?:1.8.0_191]
	... 21 more
2020-12-22T15:18:53.110Z ERROR c.a.c.DownloadableContributionsDownloader:181 [main] Cannot download the package index from https://downloads.arduino.cc/packages/package_index.json the package will be discard
java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:149) ~[arduino-core.jar:?]
	at cc.arduino.contributions.DownloadableContributionsDownloader.downloadIndexAndSignature(DownloadableContributionsDownloader.java:165) [arduino-core.jar:?]
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:306) [arduino-core.jar:?]
	at processing.app.Base.<init>(Base.java:318) [pde.jar:?]
	at processing.app.Base.main(Base.java:150) [pde.jar:?]
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: KeyUsage does not allow digital signatures
	at sun.security.ssl.Alerts.getSSLException(Alerts.java:192) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.fatal(SSLSocketImpl.java:1946) ~[?:1.8.0_191]
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:316) ~[?:1.8.0_191]
	at sun.security.ssl.Handshaker.fatalSE(Handshaker.java:310) ~[?:1.8.0_191]
	at sun.security.ssl.ClientHandshaker.serverCertificate(ClientHandshaker.java:1639) ~[?:1.8.0_191]
	at sun.security.ssl.ClientHandshaker.processMessage(ClientHandshaker.java:223) ~[?:1.8.0_191]
	at sun.security.ssl.Handshaker.processLoop(Handshaker.java:1037) ~[?:1.8.0_191]
	at sun.security.ssl.Handshaker.process_record(Handshaker.java:965) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:1064) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1367) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1395) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1379) ~[?:1.8.0_191]
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559) ~[?:1.8.0_191]
[... 80 lines cut ...]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1395) ~[?:1.8.0_191]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:1379) ~[?:1.8.0_191]
	at sun.net.www.protocol.https.HttpsClient.afterConnect(HttpsClient.java:559) ~[?:1.8.0_191]
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:185) ~[?:1.8.0_191]
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:162) ~[?:1.8.0_191]
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:155) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.HttpConnectionManager.makeConnection(HttpConnectionManager.java:106) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloaderCache.updateCacheInfo(FileDownloaderCache.java:184) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloaderCache.getFileCached(FileDownloaderCache.java:153) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:167) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129) ~[arduino-core.jar:?]
[... cut ...]
2020-12-22T15:18:53.113Z INFO c.a.c.p.ContributionInstaller:314 [main] Downloaded package index URL=[https://downloads.arduino.cc/packages/package_index.json]
2020-12-22T15:18:53.113Z INFO c.a.c.p.ContributionInstaller:324 [main] Check unknown files. Additional package index folder files=[package_index.json], Additional package index url downloaded=[]
Selected board is not available
```

I am able to open the URL using my browser without issue. 

I have tried without success:
* Hourly build
* New install
* Another connection interface (LAN / Wi-Fi)

facchinm, 2020-12-23:
Hi @bowdi ,
thanks for reporting. Is it possible that your company network is actually reencrypting SSL connections (a sort of https man in the middle proxy) ? This would explain why the certificate is not accepted (since our certificate is ok :slightly_smiling_face: ).

@cmaglie as a workaround, would it make sense to add a preference entry to avoid using https?

bowdi, 2020-12-23:
Hi @facchinm, thanks for the response!

It would appear that is the case. When I looked at the cert given, I had thought it was an original but looking closer, it was issued by Forcepoint rather than Cloudflare.

With other tools disabling SSL verification has been an option e.g. postman. The result is the same but this is where my mind went first thing ¯\_(ツ)_/¯.